OpenIntelWireless/itlwm #524: AC3165 faced scan and connect problem

Just-Prog, 2021-02-04:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: 1.3.0-DEBUG-ecf78fc (Both AirportItlwm and Itlwm)
- WiFi Card Model: AC3165
- Product ID: MacBookPro 15,3
- macOS Version: 10.14.6/10.15.7

**Description**

有时可以扫描到Wi-Fi，但无法进行连接（无法连接的Wi-Fi信号频段均为40MHz），同时在连接失败后短暂性无法继续扫描Wi-Fi。
将Wi-Fi频段切换至20mhz后均恢复正常。

**Bug Report Archive**

[AirItlDebug_2021-02-04_20-11-06.zip](https://github.com/OpenIntelWireless/itlwm/files/5925617/AirItlDebug_2021-02-04_20-11-06.zip)
[Log_2021-02-04_20-03-17.log](https://github.com/OpenIntelWireless/itlwm/files/5925618/Log_2021-02-04_20-03-17.log)

**Kext Download Source**

GitHub Releases

zxystd, 2021-02-04:
日志不完整，重新操作收集一遍吧，目测应该是跟路由器协商40mhz有问题

Just-Prog, 2021-02-05:
> 日志不完整，重新操作收集一遍吧，目测应该是跟路由器协商40mhz有问题

用dmesg重抓了一遍
[Log_2021-02-05_08-30-15.log](https://github.com/OpenIntelWireless/itlwm/files/5929343/Log_2021-02-05_08-30-15.log)

用AirItlDebug抓了60秒
[AirItlDebug_2021-02-05_08-35-05.zip](https://github.com/OpenIntelWireless/itlwm/files/5929346/AirItlDebug_2021-02-05_08-35-05.zip)

zxystd, 2021-02-05:
[归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5930949/default.zip)
@Just-Prog 试试这个。

Just-Prog, 2021-02-05:
> [归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5930949/default.zip)
> @Just-Prog 试试这个。

可以连接了，在speedtest.cn下测试速率基本在30-40Mbps之间波动，偶尔能达到50Mbps，路由器后台提供的协商速率为120。
谢谢您。

zxystd, 2021-02-05:
@Just-Prog 单天线这么差劲啊？上面显示的是40mhz吗？我回去用3165试试

Just-Prog, 2021-02-05:
> @Just-Prog 单天线这么差劲啊？上面显示的是40mhz吗？我回去用3165试试

显示的参数是“信道36，5GHz，40mHz”

kidonng, 2021-02-05:
FYI, I can get a maximum download speed of 116Mpbs with the latest commit (dbdbf007ac5bc102cd3d487b3084f5949a383978) and AC 9461.

win1010525, 2021-02-07:
> FYI, I can get a maximum download speed of 116Mpbs with the latest commit ([dbdbf00](https://github.com/OpenIntelWireless/itlwm/commit/dbdbf007ac5bc102cd3d487b3084f5949a383978)) and AC 9461.

Network is very slow because AC 3165 sucks.

zxystd, 2021-02-09:
fixed in dbdbf007ac5bc102cd3d487b3084f5949a383978.

closed 2021-02-09 by zxystd