OpenIntelWireless/itlwm #527: Heliport can't scan any wifi

James-Shadow, 2021-02-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm_v1.2.0 stable HighSierra and itlwm_1.2.0 stable 
- WiFi Card Model: AC3165
- Product ID: iMac18,1
- macOS Version: 10.13.6(17G65)

**Description**
驱动均通过Clover安装，Heliport也设置了开机启动项，但是菜单栏里没有任何Wi-Fi信息，“Wi-Fi：关闭”为灰色

**Bug Report Archive**

[终端存储的输出.log](https://github.com/OpenIntelWireless/itlwm/files/5941599/default.log)

**Kext Download Source**
GitHub Releases

win1010525, 2021-02-08:
[image]

win1010525, 2021-02-08:
Use Heliport to get logs.

James-Shadow, 2021-02-08:
> Use Heliport to get logs.

How to use it? I can't see the picture which you send

win1010525, 2021-02-08:
Please read doc first.

James-Shadow, 2021-02-08:
> Please read doc first 我通过阅读设置了boot-args，但是导出的log是空白的

James-Shadow, 2021-02-08:
> Please read doc first.

我通过阅读设置了boot-args，但是导出的log是空白的

James-Shadow, 2021-02-08:
> > Please read doc first.
> 
> 我通过阅读设置了boot-args，但是导出的log是空白的

用Heliport导出时总会报错

win1010525, 2021-02-08:
> > > Please read doc first.
> > 
> > 
> > 我通过阅读设置了boot-args，但是导出的log是空白的
> 
> 用Heliport导出时总会报错

Can you upload a screenshot of Heliport?

James-Shadow, 2021-02-08:
> > > > Please read doc first.
> > > 
> > > 
> > > 我通过阅读设置了boot-args，但是导出的log是空白的
> > 
> > 
> > 用Heliport导出时总会报错
> 
> Can you upload a screenshot of Heliport?

[bugreport_55282.zip](https://github.com/OpenIntelWireless/itlwm/files/5943272/bugreport_55282.zip)
Is this ?

James-Shadow, 2021-02-09:
> > > > > Please read doc first.
> > > > 
> > > > 
> > > > 我通过阅读设置了boot-args，但是导出的log是空白的
> > > 
> > > 
> > > 用Heliport导出时总会报错
> > 
> > 
> > Can you upload a screenshot of Heliport?
> 
> [bugreport_55282.zip](https://github.com/OpenIntelWireless/itlwm/files/5943272/bugreport_55282.zip)
> Is this ?

[image]
点击开启没反应，网络设置中也没有显示

zxystd, 2021-02-09:
1.0版本可以用吗

James-Shadow, 2021-02-09:
> 1.0版本可以用吗

不能，各个版本都试过，点开Heliportwifi是关闭的

James-Shadow, 2021-02-09:
> > 1.0版本可以用吗
> 
> 不能，各个版本都试过，点开Heliportwifi是关闭的

[image]

James-Shadow, 2021-02-09:
> > > 1.0版本可以用吗
> > 
> > 
> > 不能，各个版本都试过，点开Heliportwifi是关闭的
> 
> [image]

closed 2021-02-09 by James-Shadow