OpenIntelWireless/itlwm #528: 偶发性出现不能正常联网，查看WiFi情况，提示已连接但没有有效的ip地址

imzhengziyi, 2021-02-09:
偶发性出现不能正常联网，查看WiFi情况，提示已连接但没有有效的ip地址

基本出现在重启电脑，或者休眠后点亮屏幕后出现此情况，提示已连接，但没有有效的ip地址，获取到的IP地址不是正常的192.168.x.x号段，多次开关WiFi开关后能正常连接

James-Shadow, 2021-02-09:
[image]
你的无线网卡型号是什么，我是AC3165，为啥你会显示的那么正常😂

imzhengziyi, 2021-02-10:
我是asrock b365m itx/ac主板自带的网卡，型号我没看，我用的是最新版bigsur，看你图里还有英伟达驱动，你用的是10.13.6系统吧，应该是系统问题，我的网卡使用这个驱动，联网界面和白果网卡差不多的，没有测试别的功能！

imzhengziyi, 2021-02-10:
> [image]
> 你的无线网卡型号是什么，我是AC3165，为啥你会显示的那么正常😂

我是asrock b365m itx/ac主板自带的网卡，型号我没看，我用的是最新版bigsur，看你图里还有英伟达驱动，你用的是10.13.6系统吧，应该是系统问题，我的网卡使用这个驱动，联网界面和白果网卡差不多的，没有测试别的功能！

SeptemberHX, 2021-02-10:
点击 WiFi设置->高级->TCP/IP->DHCP续租 试试看，我在切换WIFI的时候发现没有dhcp获取新IP导致切换网络后没法联网，只能手动点击续租获取IP

imzhengziyi, 2021-02-11:
> 点击 WiFi设置->高级->TCP/IP->DHCP续租 试试看，我在切换WIFI的时候发现没有dhcp获取新IP导致切换网络后没法联网，只能手动点击续租获取IP

你说的这个方法我用过，偶尔可以解决问题，大部分情况还是不行，或者一直点，直到它正常获取IP

win1010525, 2021-02-25:
Please provide valid logs, otherwise, bat.bat will close this issue. @williambj1

williambj1, 2021-02-25:
Invalid issue format

closed 2021-02-25 by williambj1