OpenIntelWireless/itlwm #530: 你这驱动是不是挑频道？

mendax1234, 2021-02-09:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**
macOS Big sur
- Kext Version: Airporitlwm 1.2.0
- WiFi Card Model: Intel AC 9560
- Product ID: 0x8086, 0x34
- macOS Version:Big Sur

**Description**

我把自家Wi-Fi换了个5G的频道，就能正常且稳定的链接，但是用之前2.4G频道的时候就一直提示没有IP地址
**Bug Report Archive**

**Kext Download Source**

从GitHub的release上直接下载的

williambj1, 2021-02-09:
Will consider reopening once logs are properly provided.

closed 2021-02-09 by williambj1

mendax1234, 2021-02-09:
How to provide right log

发送自 Windows 10 版邮件应用

发件人: Bat.bat
发送时间: 2021年2月9日 15:27
收件人: OpenIntelWireless/itlwm
抄送: mendax1234; Author
主题: Re: [OpenIntelWireless/itlwm] 你这驱动是不是挑频道？ (#530)

Will consider reopening once logs are properly provided.

mendax1234, 2021-02-10:
> Will consider reopening once logs are properly provided.

I update my issue,

williambj1, 2021-02-10:
Did you add `msgbuf=1048576` to boot-args? According to the logs, you only tried to connect to a 5GHz network.

mendax1234, 2021-02-10:
I actually didn’t add this code , I will try it right now, thanks a lot !

发送自 Windows 10 版邮件应用

发件人: Bat.bat
发送时间: 2021年2月10日 16:02
收件人: OpenIntelWireless/itlwm
抄送: mendax1234; Author
主题: Re: [OpenIntelWireless/itlwm] 你这驱动是不是挑频道？ (#530)

Did you add msgbuf=1048576 to boot-args? According to the logs, you only tried to connect to a 5GHz network.

mendax1234, 2021-02-10:
Actually it doesn’t work at all , I still can’t connect to my 2.4G Wi-Fi, but I can’t connect to my 5G Wi-Fi

williambj1, 2021-02-10:
Please provide valid logs

mendax1234, 2021-02-10:
I have already provided the valid log!

发送自 Windows 10 版邮件应用

发件人: Bat.bat
发送时间: 2021年2月10日 20:40
收件人: OpenIntelWireless/itlwm
抄送: mendax1234; Author
主题: Re: [OpenIntelWireless/itlwm] 你这驱动是不是挑频道？ (#530)

Please provide valid logs

williambj1, 2021-02-10:
You still did not provide logs with `msgbuf=1048576`

mendax1234, 2021-02-10:
This maybe the logs with the boot-args you’ve provided.

mendax1234, 2021-02-10:
I still have another question , what’s the exact use of this code?

mendax1234, 2021-02-10:
OK， maybe this time it’s OK

发送自 Windows 10 版邮件应用

发件人: Bat.bat
发送时间: 2021年2月10日 22:00
收件人: OpenIntelWireless/itlwm
抄送: mendax1234; Author
主题: Re: [OpenIntelWireless/itlwm] 你这驱动是不是挑频道？ (#530)

still nothing

mendax1234, 2021-02-10:
[Report_2021-02-10_21-52-26.log](https://github.com/OpenIntelWireless/itlwm/files/5958622/Report_2021-02-10_21-52-26.log)
What a fxxx , the win 10 and macOS mail app is so disgusting

mendax1234, 2021-02-10:
perhaps this time it is OK

mendax1234, 2021-02-11:
> [image]
> 
> still nothing

So ， what's the problem ?