OpenIntelWireless/itlwm #531: airpotItlwm does not work after sleep, WiFi在睡眠后无法使用

ymc3, 2021-02-15:
**Have You Read Our Docs**
yes
**Are You Reporting A Bug**
yes
**Environment**

-Kext Version: airpotItlwm Mojave 1.2.0
-WiFi Card Model: Intel Dual Band Wireless AC 8265
-macOS Version: Mojave 10.14.6

**Description**
睡眠后WiFi无法使用,有图标但是没办法搜索信号.
WiFi can't be used after sleep, there is an icon but there is no way to search for signals.

**Bug Report Archive**
I don't know how to generate report, according to your http://localhost:8080/itlwm/Troubleshooting.html

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v1.2.0/AirportItlwm_v1.2.0_stable_Mojave.kext.zip

ymc3, 2021-02-15:
Wi-Fi stops working after the screen is locked and unlocked

win1010525, 2021-02-15:
No log, no solution.

ymc3, 2021-02-15:
> No log, no solution.
but how to generate log can you pls tell me

ymc3, 2021-02-15:
> 没有日志，没有解决方案。

[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/5980179/wifi.log)
is this the report that you wanted???

win1010525, 2021-02-15:
> > 没有日志，没有解决方案。
> 
> [wifi.log](https://github.com/OpenIntelWireless/itlwm/files/5980179/wifi.log)
> is this the report that you wanted???

I’m not at home. CC @williambj1

win1010525, 2021-02-15:
> > 没有日志，没有解决方案。
> 
> [wifi.log](https://github.com/OpenIntelWireless/itlwm/files/5980179/wifi.log)
> is this the report that you wanted???

This doesn't contain itlwm logs. Have you add `msgbuf=1048576` to boot-args?

ymc3, 2021-02-15:
> > > 没有日志，没有解决方案。
> > 
> > 
> > [wifi.log](https://github.com/OpenIntelWireless/itlwm/files/5980179/wifi.log)
> > is this the report that you wanted???
> 
> This doesn't contain itlwm logs. Have you add `msgbuf=1048576` to boot-args?

i did it already

win1010525, 2021-02-15:
Well, you can generate log by using heliport even if you are using airport. You can try it.

ymc3, 2021-02-15:
> Well, you can generate log by using heliport even if you are using airport. You can try it.

[bugreport_18993.zip](https://github.com/OpenIntelWireless/itlwm/files/5981504/bugreport_18993.zip)

ymc3, 2021-02-15:
the same situation when i using itlwm + heliport.

ymc3, 2021-02-15:
> > Well, you can generate log by using heliport even if you are using airport. You can try it.
> 
> [bugreport_18993.zip](https://github.com/OpenIntelWireless/itlwm/files/5981504/bugreport_18993.zip)

[bugreport_14747.zip](https://github.com/OpenIntelWireless/itlwm/files/5981547/bugreport_14747.zip)

win1010525, 2021-02-15:
failed to load init firmware
@zxystd

win1010525, 2021-02-15:
bat说你的网卡罢工了。。。你看看是不是efi哪里的pci设置不对

win1010525, 2021-02-15:
你网卡睡着之后就叫不醒了。。。

ymc3, 2021-02-15:
> bat说你的网卡罢工了。。。你看看是不是efi哪里的pci设置不对

[config.zip](https://github.com/OpenIntelWireless/itlwm/files/5984064/config.zip)

ymc3, 2021-02-15:
> > bat说你的网卡罢工了。。。你看看是不是efi哪里的pci设置不对
> 
> [config.zip](https://github.com/OpenIntelWireless/itlwm/files/5984064/config.zip)

这是我的config文件

Zero-zer0, 2021-02-16:
> > > bat说你的网卡罢工了。。。你看看是不是efi哪里的pci设置不对
> > 
> > 
> > [config.zip](https://github.com/OpenIntelWireless/itlwm/files/5984064/config.zip)
> 
> 这是我的config文件

Please remove all device properties under the path of wireless card. 'pci-aspm-default' crashes wifi after wake.

ymc3, 2021-02-16:
> > > > bat说你的网卡罢工了。。。你看看是不是efi哪里的pci设置不对
> > > 
> > > 
> > > [config.zip](https://github.com/OpenIntelWireless/itlwm/files/5984064/config.zip)
> > 
> > 
> > 这是我的config文件
> 
> Please remove all device properties under the path of wireless card. `pci-aspm-default` crashes wifi after wake.

thanks for you method, the problem finally fixed.
thank you so much !!!

closed 2021-02-16 by ymc3