OpenIntelWireless/itlwm #535: Airdrop doesn't work

Yaya48, 2021-02-22:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: Airportitlwm 1.2
- WiFi Card Model: Intel AX200 160MHz
- Product ID: AX200NGW
- macOS Version: 11.2.1

**Description**

Airdrop does not send/receive file however he does show the phone in Mac OS finder and sometime on the phone itself.

The kext has been downloaded from here.
Report log here: 
[Report_2021-02-22_23-14-22.log](https://github.com/OpenIntelWireless/itlwm/files/6025509/Report_2021-02-22_23-14-22.log)

mehmetyukcell, 2021-02-22:
AirDrop not Supported, now. May be after

Yaya48, 2021-02-22:
> AirDrop not Supported, now. May be after 

Ah thought it was make sense thank

williambj1, 2021-02-25:
Duplicate of #292

closed 2021-02-25 by williambj1