OpenIntelWireless/itlwm #536: Wifi Cant Work after wakeup, only solution is reboot

fangfang94, 2021-02-24:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.2.0 First, and change to 1.3.0 still no luck. + Heliport
- WiFi Card Model: intel wireless 8265 / 8275 
- macOS Version: Catalina 10.15.7

**Description**
WiFi can't be used after sleep, still appear after wake up but cant turn on and search

**Bug Report Archive**

[bugreport_26956.zip](https://github.com/OpenIntelWireless/itlwm/files/6032691/bugreport_26956.zip)

**Kext Download Source**

Release

fangfang94, 2021-02-24:
I am new to hackintosh, please forgive me if I provide anything wrong

williambj1, 2021-02-24:
You should remove incompatible PCI properties, especially ones that break aspm

```
			<key>PciRoot(0x0)/Pci(0x1D,0x0)/Pci(0x0,0x0)</key>
			<dict>
				<key>AAPL,slot-name</key>
				<string>WLAN</string>
				<key>compatible</key>
				<string>pci14e4,43a0</string>
				<key>device_type</key>
				<string>Airport Extreme</string>
				<key>model</key>
				<string>BCM4352 802.11ac Wireless Network Adapter</string>
				<key>name</key>
				<string>Airport</string>
				<key>pci-aspm-default</key>
				<integer>0</integer>
			</dict>
```

closed and locked as invalid

closed 2021-02-24 by williambj1