OpenIntelWireless/itlwm #537: 总感觉蓝牙和WIFI驱动有冲突！

llaisyuan, 2021-02-25:
不知道我发的会不会有人去解决或观注
现在11G系统OC6.6 网卡AX200
我的情况比较特殊：WIFI都能正常使用，蓝牙不管处于什么状态都不工作
1：WIFI和蓝牙驱动都在OC中加载，那么 WIFI正常使用 蓝牙在USB中看不到，且蓝牙设备属性为空
2: 只加载WIFI驱动，那么USB中不会出现蓝牙，但蓝牙属性中会看到蓝牙这个设备，厂商识别为爱立信
3：不管任何状态，放入蓝牙驱动，那么USB中和蓝牙属性中都看不到蓝牙设备
4：如果驱动都不放，那么蓝牙在USB中会出现，且蓝牙设备管理中可以看到它
USB端口我也定置过，我在没有WIFI的状态下，能看到蓝牙USB端口的时候去定置了，然后重新加载WIFI和蓝牙驱动也没有用
请求帮忙，或给出建议

QQ：973264

win1010525, 2021-02-25:
Invalid as you never load IntelBluetoothFirmware normally.

raydoom, 2021-02-25:
我的7260也有类似的问题，系统是11.2.1，OC0.6.3，AirportItlwm v1.3.0-alpha
1，wifi每隔十几个小时就会自动断线，WiFi图标显示连接正常，就是网络不通，重新手动连接可以恢复
2，放入蓝牙驱动，USB中和蓝牙属性中都看不到蓝牙设备，以前在v1.1.0 stable的时候蓝牙是可以使用的

llaisyuan, 2021-02-25:
放入蓝牙驱动，USB中和蓝牙属性中都看不到蓝牙设备，以前在v1.1.0 stable的时候蓝牙是可以使用的
你这一条。和我的特别像。。不放蓝牙驱动。还是能看到蓝牙的硬件属性

llaisyuan, 2021-02-25:
第一条，我有时也断网，我换了蓝牙1.3驱动。有改善。

williambj1, 2021-02-25:
Do not spam issues everywhere

closed 2021-02-25 by williambj1