OpenIntelWireless/itlwm #547: 42bef24 commit  AC8260 not getting ip adress after reconect or connect to other wifi

mhl221135, 2021-03-14:
**Have You Read Our Docs**
Yes
**Are You Reporting A Bug**
Yes 
**Environment**

- Kext Version: 42bef24 commit   
- WiFi Card Model: AC8260
- Product ID: ?
- macOS Version: 11.2.2

**Description**
42bef24 commit  AC8260 not getting ip adress after reconect or connect to other wifi

**Bug Report Archive**
kextstat | grep -i "airportitlwm" 
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
   83    0 0                  0x1d1000   0x1d1000   com.zxystd.AirportItlwm (1.3.0) 84692218-8B1F-3453-901E-5195381BB3F4 <82 46 14 6 5 3 1>

----- Kext Status -----

   83    0 0                  0x1d1000   0x1d1000   com.zxystd.AirportItlwm (1.3.0) 84692218-8B1F-3453-901E-5195381BB3F4 <82 46 14 6 5 3 1>

----- Airport Info -----

     agrCtlRSSI: -53
     agrExtRSSI: 0
    agrCtlNoise: -88
    agrExtNoise: 0
          state: running
        op mode: station 
     lastTxRate: 300
        maxRate: 300
lastAssocStatus: 0
    802.11 auth: open
      link auth: wpa2-psk
            MCS: 15
        channel: 149,-1

----- Airport Sys Report -----

Wi-Fi:

      Software Versions:
          CoreWLAN: 16.0 (1657)
          CoreWLANKit: 16.0 (1657)
          Menu Extra: 17.0 (1728)
          System Information: 15.0 (1502)
          IO80211 Family: 12.0 (1200.12.2b1)
          Diagnostics: 11.0 (1163)
          AirPort Utility: 6.3.9 (639.15)
      Interfaces:
        en1:
          Card Type: AirPort Extreme  (0x8086, 0x130)
          Firmware Version: itlwm: 1.3.0 fw: 34.0.1
          Locale: FCC
          Country Code: ZZ
          Supported PHY Modes: 802.11 a/b/g/n/ac
          Supported Channels: 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149, 153, 157, 161, 165
          Wake On Wireless: Supported
          AirDrop: Supported
          AirDrop Channel: 149
          Auto Unlock: Supported
          Status: Not Associated

----- Bluetooth Sys Report -----

Bluetooth:

      Apple Bluetooth Software Version: 8.0.3d9
      Hardware, Features, and Settings:
          Bluetooth Low Energy Supported: Yes
          Handoff Supported: Yes
          Instant Hot Spot Supported: Yes
          Manufacturer: Intel
          Transport: USB
          Firmware Version: 7.256 (7.256)
          Connectable: Yes
          Auto Seek Pointing: On
          Remote wake: Off
          Vendor ID: 0x8087
          Product ID: 0x0A2B
          Auto Seek Keyboard: On
      Devices (Paired, Configured, etc.):
          Device 0:
              Major Type: Peripheral
              Minor Type: Mouse
              Services: YiChip Bluetooth Mouse
              Manufacturer: Cambridge Silicon Radio (0x6, 0x1D87)
              Bluetooth Core Spec: 4,0
              Firmware Version: 0x0129
              Vendor ID: 0x046D
              Product ID: 0xB016
              Host Connectable: No
              EDR Supported: No
              eSCO Supported: No
              SSP Supported: Yes
          Device 1:
              Major Type: Audio
              Minor Type: Headset
              Services: AVRCP TG, Audio SNK, Hands-free unit, AVRCP CT, Headset unit
              Manufacturer: Unknown (0x7, 0x305)
              Bluetooth Core Spec: 4,1
              Firmware Version: 0x0305
              Vendor ID: 0x0039
              Product ID: 0x1582
              EDR Supported: Yes
              eSCO Supported: Yes
              SSP Supported: Yes
          Device 2:
              Major Type: Audio
              Minor Type: Headset
              Services: Handsfree Unit, A2DP SNK Service, Human Interface Device, A/V Remote Control
              Bluetooth Core Spec: 4,2
              EDR Supported: Yes
              eSCO Supported: Yes
              SSP Supported: Yes

**Kext Download Source**
downloaded from itlwm git and removed other cards firmawares except iwm-8000C*

williambj1, 2021-03-14:
Follow [this guide](https://openintelwireless.github.io/itlwm/Troubleshooting.html#using-dmesg) and attach logs. The given info is not useful for debugging. This issue will be reopened once the logs are correctly provided.

closed 2021-03-14 by williambj1

mhl221135, 2021-03-14:
Logs from asked guide 
[Log_2021-03-14_17-15-44.log](https://github.com/OpenIntelWireless/itlwm/files/6136943/Log_2021-03-14_17-15-44.log)