OpenIntelWireless/itlwm #549: 网络卡死后无法切换网络  Wifi menu frozen after network stuck.

howie95, 2021-03-19:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm 1.2.0 release & 1.3.0 alpha
- WiFi Card Model:  AC 9560 (9560d2w)
- Product ID: -
- macOS Version:  Catalina 10.15.6

**Description**
- 刚开机时是正常的，一段时间使用后有机率网络会卡死，并且此时在Wifi列表里点击其他网络切换的话没有任何反应，并且这时候打开系统设置-网络面板，里面显示是空的，过一会才能加载出来Wifi和蓝牙选项。这时候再切换另一个已知网络，大概率过一会就提示连接失败要输入密码。
- When restart the Wifi works fine, after a period of use, there is a chance that the network will be stuck, and at this time in the Wifi-select list, click on other networks to switch is no response, then when open the system settings - network panel, there shows nothing, have to wait for a while to see the Wifi and Bluetooth options. After that when switch to another saved network, much probability is connection failure and require to enter the password again.

**Bug Report Archive**
- Log using dmesg
[Log_2021-03-19_18-33-52.log](https://github.com/OpenIntelWireless/itlwm/files/6171054/Log_2021-03-19_18-33-52.log)
- Runtime Logs (not sure what is operation time period, just set it to 20 seconds)
[Report_2021-03-19_19-56-50.log](https://github.com/OpenIntelWireless/itlwm/files/6171063/Report_2021-03-19_19-56-50.log)
- System Wifi Logs (I think the trouble may happend around 18:00)
[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/6171070/wifi.log)

Please have a look, thanks!

williambj1, 2021-03-19:
Looks like you are spamming the connect button at a ridiculously high frequency, could you attach another log without such behavior?

williambj1, 2021-03-19:
dmesg log is sufficient for debugging, please try itlwm instead of AirportItlwm as well

howie95, 2021-03-22:
> dmesg log is sufficient for debugging, please try itlwm instead of AirportItlwm as well

Thanks, i'll try it later.
I have set of machines that have same hardware, but it sames like only this one has the problem, and its hard to reproduce the problem cause i'm not the main user of that machine and i'm not sure what actually happend when the problem appeared :(

howie95, 2021-03-29:
Since the other machines with same hardware didn't produce the same problem, i think maybe the problem is caused by the specific hardware or WiFi, i'll close this issue for now.

closed 2021-03-29 by howie95