OpenIntelWireless/itlwm #550: [AirportITLWM] Wi-Fi works but shows up as Ethernet

martibravo, 2021-03-19:
AirportItlwm was working fine, even Airdrop was kind of working. Now, when I start my computer it connects to my home Wi-Fi, but it shows up as Ethernet on Network settings panel and I can't switch networks. 

**System:**
Laptop: Matebook D 14 
Processor: Intel i5 8250u
OS: Big Sur 11.2
RAM: 8gb
Wifi Card: Intel AC8625

[image]

r33int, 2021-03-19:
Try resetting your network interfaces:
`sudo rm /Library/Preferences/SystemConfiguration/NetworkInterfaces.plist
sudo rm /Library/Preferences/SystemConfiguration/preferences.plist`

williambj1, 2021-03-19:
Read our docs and reset your network interface preferences.

closed 2021-03-19 by williambj1

martibravo, 2021-03-22:
It was my fault, I updated OpenCore and forgot to delete itlwm and leave AirportItlwm only. Sorry!