OpenIntelWireless/itlwm #552: Wifi and bluetooth not connect same time

nhatminhhp1999, 2021-03-23:
I think the problem bluetooth can not use at the same time with wifi the main problem is due to the 5ghz wave frequency and the 80mhz 40mhz band. Supporting these frequencies in the future will completely fix the bluetooth problem

sojab0on, 2021-03-24:
only issue I found with connections on Bluetooth is mice or keyboards my love 101 , jabra t110bt and JBL care 2 speakers connect without issue to my intel Bluetooth from the intel 9560AC 160mhz card

rboldini, 2021-07-12:
Actually this issue is found on 7XXX and 8XXX series, with my 7265AC (2 antennas), speakers with low audio codec (SBC I think) works with wifi connected but with a low range distance from laptop, devices like AirPods or newer headphones don't work .

fredimartins, 2021-10-01:
i have issues with bluetooth and wifi at same time, bluetooth headphones are popping audio if the headphones are more than 20cm. When i disable wifi, bluetooth audio stay smoothly even i stay 2 meters from notebook.

williambj1, 2021-10-02:
I don't think that's the case, might be related to coex configs with Gen 1 Bluetooth but still not sure at this point. https://github.com/OpenIntelWireless/IntelBluetoothFirmware/blob/master/IntelBluetoothFirmware/IntelBluetoothFirmware.cpp#L152

The range issue discussion can be continued in #692

closed 2021-10-02 by williambj1