OpenIntelWireless/itlwm #554: ax200频繁断开和重连

thep0y, 2021-03-25:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.3.0 alpha
- WiFi Card Model: ax200
- Product ID: 不知道哪里获取
- macOS Version: 11.2.3

**Description**
[image]

断开和重连发生时，可以通过手动点击Wi-Fi名称解决，但过一段时间会再次发生。

**Bug Report Archive**
[AirItlDebug_2021-03-25_15-34-16.zip](https://github.com/OpenIntelWireless/itlwm/files/6202886/AirItlDebug_2021-03-25_15-34-16.zip)

**Kext Download Source**
最新release中下载

778280151, 2021-03-25:
一样的网卡，同样的问题，不论怎么更改驱动顺序，在不休眠黑屏的情况下，突然就没网了

thep0y, 2021-03-26:
> 一样的网卡，同样的问题，不论怎么更改驱动顺序，在不休眠黑屏的情况下，突然就没网了

你可以换路由器试一下，我自己觉得可能与路由器有关，之前用红米ax5的时候这种情况没遇到过，换了华硕后就出现了。
但不一定和路由器有关，现在某东可以试用7天，试一下红米ax6没准能解决。

778280151, 2021-03-27:
> > 一样的网卡，同样的问题，不论怎么更改驱动顺序，在不休眠黑屏的情况下，突然就没网了
> 
> 你可以换路由器试一下，我自己觉得可能与路由器有关，之前用红米ax5的时候这种情况没遇到过，换了华硕后就出现了。
> 但不一定和路由器有关，现在某东可以试用7天，试一下红米ax6没准能解决。
不行，我家里本来就是ax6和ax5组网，这种偶发性问题真的难受

thep0y, 2021-03-28:
> 不行，我家里本来就是ax6和ax5组网，这种偶发性问题真的难受
今天又换回到红米ax6，除了唤醒后需要等待一段时间连接，暂时没发现有断连的问题。

778280151, 2021-03-28:
> 今天又换回到红米ax6，除了唤醒后需要等待一段时间连接，暂时没发现有断连的问题。
没办法了，老哥你隔空传送能用不？我这边能发现但是不能传

thep0y, 2021-03-29:
> 今天又换回到红米ax6，除了唤醒后需要等待一段时间连接，暂时没发现有断连的问题。

没用过这个功能，手里也没有iphone测试，过段时间买个12pm试一下

EGOISTK21, 2021-04-07:
我的ax86配合ax200也是时常断连

778280151, 2021-04-09:
> 我的ax86配合ax200也是时常断连

我的不是ax86 我是小米的ax5和ax6组网，都不行，有机会拿去公司试一下，但是感觉还是会断，和路由器关系不大

zxystd, 2021-04-10:
有没有试过itlwm.kext? 另外不提供日志的话我只能算卦了

kingo132, 2021-04-26:
我的ax200在某个路由器下也是频繁重启，可能是驱动崩溃了，我没留意看日志。
当然我在其它大部分路由器下面，都不会出现频繁重启。所以我猜测，可能还是某些特定的路由器配置会导致这个问题。如果你只是想不频繁重启，建议修改路由器设置。
如果要解决这个bug，还是需要一些详细的日志给作者来分析这个问题。下次我再遇到这个问题了就提交些日志看看。

778280151, 2021-04-27:
> 有没有试过itlwm.kext? 另外不提供日志的话我只能算卦了

老哥，怎么获取debug日志呀

778280151, 2021-04-27:
> 我的ax200在某个路由器下也是频繁重启，可能是驱动崩溃了，我没留意看日志。
> 当然我在其它大部分路由器下面，都不会出现频繁重启。所以我猜测，可能还是某些特定的路由器配置会导致这个问题。如果你只是想不频繁重启，建议修改路由器设置。
> 如果要解决这个bug，还是需要一些详细的日志给作者来分析这个问题。下次我再遇到这个问题了就提交些日志看看。

我目前用的是最新的2.0，现在电脑拿到公司了，发现貌似和老的wifi协议好像冲突，不能正常分配ip，顺便问下，怎么捕捉debug日志😂

kingo132, 2021-04-27:
> > 我的ax200在某个路由器下也是频繁重启，可能是驱动崩溃了，我没留意看日志。
> > 当然我在其它大部分路由器下面，都不会出现频繁重启。所以我猜测，可能还是某些特定的路由器配置会导致这个问题。如果你只是想不频繁重启，建议修改路由器设置。
> > 如果要解决这个bug，还是需要一些详细的日志给作者来分析这个问题。下次我再遇到这个问题了就提交些日志看看。
> 
> 我目前用的是最新的2.0，现在电脑拿到公司了，发现貌似和老的wifi协议好像冲突，不能正常分配ip，顺便问下，怎么捕捉debug日志😂

启动参数加上
```
keepsyms=1 debug=0x100 msgbuf=1048576
```
然后开机以后执行
```
sudo dmesg | grep -i itlwm
```

778280151, 2021-04-28:
> 启动参数加上
>
> ```
> keepsyms=1 debug=0x100 msgbuf=1048576
> ```
>
> 然后开机以后执行
>
> ```
> sudo dmesg | grep -i itlwm
> ```

ok,谢谢了

wqhqq1, 2021-04-29:
华为ax3看起来还好
没怎么断
但就是有时候连不上160mhz，要重启路由器才行

zxystd, 2021-05-07:
有没有试过最新的版本？另外如果使用的是alpha版本的话请带上提交的hash，不然我不知道你用的到底是哪个。

win1010525, 2021-07-06:
Duplicate of #518

williambj1, 2021-08-17:
Closed since this issue is no longer active. Feel free to continue posting comments if you are still experiencing this issue.

closed 2021-08-17 by williambj1