OpenIntelWireless/itlwm #555: Didn't load AirportItlwm.kext on Catalina

ducltv, 2021-03-28:
Hi guys. I have some issues with AirportItlwm kext on catalina, i have put that in the OC/Kexts/ and add it in config but when i turn on the pc it doesn't load the Airportitlwm althrough the IO80211Family loaded
[image]

ducltv, 2021-03-28:
I confirmed i save the config before restart the pc.

ducltv, 2021-03-28:
[image]
Well after i change something in the config it work :l I don't know how the config can make me confused about 30 mins.

closed 2021-03-28 by ducltv

Malumen, 2021-04-05:
> [image]
> Well after **i change something in the config** it work :l I don't know how the config can make me confused about 30 mins.

What did you change? I am also having trouble getting it to work!

zcream, 2021-06-23:
I've got a similar problem. Do let us know what you changed?

ducltv, 2021-08-03:
@zcream i remember it about the kind of arch each kext i guess

joshrogershak, 2022-08-19:
please tell us what you changed exactly - i rlly need it for something @fishudroid