OpenIntelWireless/itlwm #564: Airportitlwm is loaded but unable to find networks

Minipomy, 2021-04-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm_v1.2.0_stable_BigSur.kext
- WiFi Card Model: Intel Dual-Band Wireless-AC 7260
- Product ID: Vendor: 8086 | Device: 08B1
- macOS Version: 11.2.3 Big Sur (20D91)

**Description**
Airportitlwm.kext is unable to shows available networks, i followed the recommended settings like AppleSecureModel and therefore like forcing IO80211Family.kext but i still without luck, could you help me?

**Bug Report Archive**

DMESG: [Log_2021-04-07_21-56-32.log](https://github.com/OpenIntelWireless/itlwm/files/6276146/Log_2021-04-07_21-56-32.log)

RUNTIME LOGS: [Report_2021-04-07_21-58-56.log](https://github.com/OpenIntelWireless/itlwm/files/6276147/Report_2021-04-07_21-58-56.log)

**Kext Download Source**
This repo

williambj1, 2021-04-08:
Looks like both IO80211Family and AirportItlwm loaded properly but IO80211Family failed to request AirportItlwm to scan networks. Could you please try the following procedures to help us clarify the issue?

1. Try resetting network interface preferences (instructions can be found in our docs)
2. Test if AirportItlwm works in macOS Recovery
3. Try itlwm with HeliPort

zxystd, 2021-04-09:
please fix your dsdt.

closed 2021-04-09 by zxystd

Minipomy, 2021-04-09:
> Looks like both IO80211Family and AirportItlwm loaded properly but IO80211Family failed to request AirportItlwm to scan networks. Could you please try the following procedures to help us clarify the [...]

Hi Williambj1, thanks for the fast reply.

AirportItlwm.kext has the same symptom in "installer", i also switched just today to itlwm+heliport with no success... the machine is fresh installed with the minium detail that "SecureBootModel" is now set as "Disable"

> please fix your dsdt.

i was thinking of that, because my usb ports are not yet mapped... could be that related? i'm using usbinject.kext

i'm open suggestions!

P.D.: i'll now try in recovery mode AirportItlwm with SecureBootModel as "Default"

And as always, thanks so much for the patience and the fast reply!

Minipomy, 2021-04-11:
Hi Williambj1!

After a long road i managed to fix the problem.

after making a fresh install and making all my DSDTs/SSDTs related patches and gathering all the kexts that i need, i was thinking of the "airplane mode" of the laptop because i have no indicator of it (shameless), after playing around with a few buttons on my keyboard and do the mapping of it i manage to turn on my Intel AC 7260 and the Bluetooth!

Now just wondering... could be that airportitlwm has some issues connecting to 5GHz networks?

As always thanks for the amazing job, take care!

Best regards!

maobui, 2021-08-12:
Thank you! I lost a few days to realize that I'm in airplane mode.

maobui, 2021-08-12:
> Hi Williambj1!
>
> After a long road i managed to fix the problem.
>
> after making a fresh install and making all my DSDTs/SSDTs related patches and gathering all the kexts that i need, i was thinking of [...]

Thank you! I lost a few days to realize that I'm in airplane mode.

jhonh10, 2022-02-16:
hi, please can help me. i have same wifi card but don't search wifi networks, how did you solve it?

ahaolee, 2023-03-11:
> Hi Williambj1!
>
> After a long road i managed to fix the problem.
>
> after making a fresh install and making all my DSDTs/SSDTs related patches and gathering all the kexts that i need, i was thinking of [...]

thank you so much , I search many many information in Internet and github , and i reboot ~100 times , resetnvram 100times, and reinstall macos down to montery...   wifi adapter still cannot scan ssid .  untill looked your  encounter . I click this airplane mode keyboard button , wifi adapter connecting my wifi . thank you again !!!

mahmudunnabiRain, 2024-03-09:
> Hi Williambj1!
>
> After a long road i managed to fix the problem.
>
> after making a fresh install and making all my DSDTs/SSDTs related patches and gathering all the kexts that i need, i was thinking of [...]

Saved my day. I didn't know my laptop has "airplane mode" . I have a dell laptop which has a shortcut function key to enable "airplane mode" I pressed and all wifi network kept showing up. thanks man.