OpenIntelWireless/itlwm #566: bluetooth range and wifi download speed

priyanshuarora595, 2021-04-12:
bluetooth range is very low... I can't even walk away from my desk , the bluetooth gets disconnected . bluetooth range is merely 3-4 feet.

other issue is that download speed can't be monitored by using istat menus. it always shows 0 download speed. I am using Intel(R) Dual Band Wireless AC 3165.  

please fix it.

thank you

williambj1, 2021-04-13:
1. 3k series are very low end single antenna cards, consider replacing it if you want a better bluetooth experience.
2. Fixed months ago

Closed since the issue format is invalid.

closed 2021-04-13 by williambj1

priyanshuarora595, 2021-04-13:
> 1. 3k series are very low end single antenna cards, consider replacing it if you want a better bluetooth experience.
> 2. Fixed months ago
> 
> Closed since the issue format is invalid.

In Windows it jist works great , no issue , then why in hackintosh . ?

williambj1, 2021-04-13:
The is no proper support for Intel Bluetooth right now (read the docs for more info); the lack of Coex configuration might affect the performance as well. We don't have plans to make any improvements in the near future, feel free to submit a pull request.

priyanshuarora595, 2021-04-13:
> The is no proper support for Intel Bluetooth right now (read the docs for more info); the lack of Coex configuration might affect the performance as well. We don't have plans to make any improvements in the near future, feel free to submit a pull request.

Okay thank you