OpenIntelWireless/itlwm #567: AirportItlwm不能连接WPA2 Enterprise

fanksy, 2021-04-12:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes 

**Environment**

- Kext Version: 1.3.0 alpha
- WiFi Card Model: intel 7260n
- Product ID: 0x8086 0x08B2
- macOS Version: macOS 10.15.7

**Description**
[image]

**Bug Report Archive**

[Report_2021-04-12_17-41-15.log](https://github.com/OpenIntelWireless/itlwm/files/6295636/Report_2021-04-12_17-41-15.log)

**Kext Download Source**
github release

soulmercy, 2021-04-12:
Same problem with AirportItlwm on AX200

CCA2878, 2021-04-14:
Same problem with AirportItlwm on AX201.
Also cannot connect WPA2 Enterprise WiFi in Windows10.

RaymdZ, 2021-04-15:
It works here with AirportItlwm 1.3, AC9560, and Big Sur 11.2. 
The only issue is that manual DHCP update is mostly required, and connection loses easily when signal is weak. 
[image]

fanksy, 2021-05-08:
升级到10.15.7(19H1030)版本后能连接WPA2 Enterprise
操作过程：
直接从wifi列表点击并没有直接弹出账号输入框，需要点击“Join Other Network...”后输入wifi名称选择“WPA2 Enterprise”进行连接，输入账号后点击“Join”后弹出了信任证书的操作，这一步在升级之前是没有的，确定后就直接连接成功了 
@soulmercy @RaymdZ @CCA2878

CCA2878, 2021-05-08:
> 升级到10.15.7(19H1030)版本后能连接WPA2 Enterprise
> 操作过程：
> 直接从wifi列表点击并没有直接弹出账号输入框，需要点击“Join Other Network...”后输入wifi名称选择“WPA2 Enterprise”进行连接，输入账号点击“Join”后弹出了信任证书的操作，这一步在升级之前是没有的，确定后就直接连接成功了
> @soulmercy @RaymdZ @CCA2878

11.3 BigSur 仍然不行，另外确实Win10下也不行

fanksy, 2021-05-08:
> > 升级到10.15.7(19H1030)版本后能连接WPA2 Enterprise
> > 操作过程：
> > 直接从wifi列表点击并没有直接弹出账号输入框，需要点击“Join Other Network...”后输入wifi名称选择“WPA2 Enterprise”进行连接，输入账号点击“Join”后弹出了信任证书的操作，这一步在升级之前是没有的，确定后就直接连接成功了
> > @soulmercy @RaymdZ @CCA2878
> 
> 11.3 BigSur 仍然不行，另外确实Win10下也不行

https://openintelwireless.github.io/itlwm/Troubleshooting.html#resetting-system-preferences-network
试试这个

win1010525, 2021-05-24:
Same on AC8260, Big Sur 11.3.

closed 2021-06-04 by zxystd