OpenIntelWireless/itlwm #570: AirportItlwm断流

EntityBug, 2021-04-15:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**
AC9462 macOS 11.2.3 AirportItlwm1.3.0

- Kext Version: 1.3.0
- WiFi Card Model: AC9462
- Product ID: 找不到
- macOS Version: 11.2.3

**Description**

存在两种断流
1.系统显示正常，但无法访问任何网站，一般重新开启可在一段时间内稳定连接
2.系统显示正常，但无法访问任何网站，App在关闭Wi-Fi后仍会认为有网络连接（微信、qq不显示未联网，Chrome尝试加载网页），重新连接需要较长时间，且只能保持不到1分钟的连接，一般发生在19:00到21:00（至少在17:00以后是这样）

**Bug Report Archive**

[Report_2021-04-15_18-38-43.log](https://github.com/OpenIntelWireless/itlwm/files/6318251/Report_2021-04-15_18-38-43.log)

**Kext Download Source**

[AirportItlwm_1.3.0_bigsur.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6318257/AirportItlwm_1.3.0_bigsur.kext.zip)

win1010525, 2021-04-20:
你应该研究一下驱动是如何做到分时间断流的

zxystd, 2021-05-07:
有没有试过2.0.0版本？

EntityBug, 2021-05-08:
> 有没有试过2.0.0版本？

试过，问题仍存在

joaovictorsouza, 2021-05-17:
Same problem here with both versions described.

williambj1, 2021-10-02:
@EntityBug Have you tired `itlwm.kext`?

EntityBug, 2021-10-02:
> @EntityBug Have you tired `itlwm.kext`?

Yes.

zxystd, 2022-05-08:
Use the latest one.

closed 2022-05-08 by zxystd