OpenIntelWireless/itlwm #572: AX200 unable to reach fullspeed on Internet but on Intranet > 100Mbps | AX200 Speedtest到ISP服务器无法满速，可是内网传输>100Mbps

samleong123, 2021-04-18:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: 1.3.0 Alpha
- WiFi Card Model: Intel AX201
- Product ID: I don't know where to get it. 我不知道哪里可以找到
- macOS Version: Big Sur 11.2.3

**Description**

Tested on internal network connected to my router through 5GHz , able to success reaching 14.6Mb/s (14.6Mb/s * 8 = 116.8 Mbps)
My house internet package is 50Mbps Downloads and 20Mbps upload. Tested with no device connected to the router. Im only getting < 20Mbps on both DL and UL.

连接到我路由器的5GHz，内网传输下载可以达到23MB/s+ (184Mbps) ，上传可以达到16Mb/s (128Mbps)
我家ISP给的是 50DL 20UL ，可是Speedtest却跑不满50DL 20UL，可是内网随便都可以上>100Mbps

内网下载：
[image]

内网上传：
[image]

ISP服务器测试：
[image]
Speedtest的时候后台无任何占用网络的资源/APP

**Bug Report Archive**

[Log_2021-04-18_12-04-18.log](https://github.com/OpenIntelWireless/itlwm/files/6330476/Log_2021-04-18_12-04-18.log)

**Kext Download Source**

Github Release on https://github.com/OpenIntelWireless/itlwm/

williambj1, 2021-04-18:
Contact your ISP, this is completely unrelated to this kext.

closed 2021-04-18 by williambj1

samleong123, 2021-04-18:
but tested on ethernet full speed reached
50DL 20UL with ethernet but not wifi 5ghz
tested on AX201 (another devices) full speed with 5GHz
[image]

zxystd, 2021-04-18:
老哥你拿有线对比无线真的好么？