OpenIntelWireless/itlwm #575: 无法连接WPA/WPA2 PSK

Haaooo, 2021-04-20:
**Have You Read Our Docs**

YES
**Are You Reporting A Bug**

YES
**Environment**

- Kext Version: AirportItlwm_v1.3.0_stable_BigSur.kext
- WiFi Card Model: intel ac 9462
- Product ID: 
- macOS Version:11.3 Beta版(20E5186e)
- **Description**

**Bug Report Archive**

**Kext Download Source**

zxystd, 2021-04-20:
log?

Haaooo, 2021-04-20:
> 日志？

我是小白，不知道在哪找日志

win1010525, 2021-04-20:
小白就不要来开issue了

williambj1, 2021-04-20:
Will consider reopening if logs are correctly provided.

closed 2021-04-20 by williambj1

Haaooo, 2021-04-20:
> 如果日志提供正确，将考虑重新开放。I have uploaded the log, but I don't know if it is this one

Haaooo, 2021-04-20:
> 如果日志提供正确，将考虑重新开放。
I have uploaded the log, but I don't know if it is this one

zxystd, 2021-04-20:
日志不对，你要按照文档里面说的来，加了bootargs之后用heliport来收集日志也可以

Haaooo, 2021-04-28:
> 日志不对，你按照文档说的，用直升机收集日志后，也可以

[Uploading AirItlDebug_2021-04-28_22-14-07.zip…]()