OpenIntelWireless/itlwm #580: 2.0.0 ax200 无法连接160mhz Wi-Fi

wqhqq1, 2021-04-24:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: 2.0.0
- WiFi Card Model: ax200
- Product ID: none
- macOS Version: 11.2.3

**Description**

几天前还能连接到160mhz，今天突然不行了，只有80mhz，用的是最新的alpha commit b411ca9d2ab1504d205fd97da44e540b7f13009b
**Bug Report Archive**

日志稍后传
**Kext Download Source**

GitHub release b411ca9d2ab1504d205fd97da44e540b7f13009b

zxystd, 2021-04-24:
重启路由器

closed 2021-04-24 by zxystd

wqhqq1, 2021-04-24:
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf5ba70> [channelNumber=40(5GHz), channelWidth={80MHz160MHz}],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf59110> [channelNumber=44(5GHz), channelWidth={80MHz160MHz}],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf3e410> [channelNumber=48(5GHz), channelWidth={80MHz160MHz}],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf71520> [channelNumber=52(5GHz), channelWidth={80MHz160MHz}]
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf42650> [channelNumber=56(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf56a10> [channelNumber=60(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf52a10> [channelNumber=64(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf6d9a0> [channelNumber=100(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf6beb0> [channelNumber=104(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf6b950> [channelNumber=108(5GHz), channelWidth={80MHz160MHz}, DFS]
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf5a450> [channelNumber=112(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf53600> [channelNumber=116(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf50070> [channelNumber=120(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf4eb60> [channelNumber=124(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf50790> [channelNumber=128(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:46:     <CWChannel: 0x7fb6dbf2e3b0> [channelNumber=132(5GHz), channelWidth={80MHz160MHz}, DFS]
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc89cd0> [channelNumber=40(5GHz), channelWidth={80MHz160MHz}],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc816f0> [channelNumber=44(5GHz), channelWidth={80MHz160MHz}],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc81790> [channelNumber=48(5GHz), channelWidth={80MHz160MHz}],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc4f400> [channelNumber=52(5GHz), channelWidth={80MHz160MHz}]
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc7b1d0> [channelNumber=56(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc7b250> [channelNumber=60(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc4e4e0> [channelNumber=64(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc49950> [channelNumber=100(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc49980> [channelNumber=104(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc49990> [channelNumber=108(5GHz), channelWidth={80MHz160MHz}, DFS]
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc89d00> [channelNumber=112(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc81760> [channelNumber=116(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc453d0> [channelNumber=120(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc4f410> [channelNumber=124(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc7b240> [channelNumber=128(5GHz), channelWidth={80MHz160MHz}, DFS],
Sat Apr 24 16:49:53:     <CWChannel: 0x7fb6dbc4e470> [channelNumber=132(5GHz), channelWidth={80MHz160MHz}, DFS]
2021-04-24 16:46:34.919986+0800 0xbfc      Activity    0x3160               408    0    sharingd: Sharing/SDActivityAdvertiser/continuityDidStopAdvertisingOfType
2021-04-24 16:46:37.622347+0800 0xcbe      Default     0x0                  408    3    sharingd: [com.apple.sharing:SDAirDropHashStore] Change anchor: <CNChangeHistoryAnchor: 0x7fe773cc9160: token=<NSPersistentHistoryToken - {
2021-04-24 16:49:46.358160+0800 0x165      Default     0x0                  0      0    kernel: (IO80211Family) IO80211ScanManager::scanDone: Scheduling cache purge timer in 30 seconds.
2021-04-24 16:49:50.252160+0800 0x33c      Default     0x0                  0      0    kernel: (IO80211Family) 0,
2021-04-24 16:50:16.375404+0800 0x0        Default     0x0                  0      0    itlwm: ieee80211_vht_negotiate 804 chan_width=3 support_160=1 support_80_80=1
2021-04-24 16:50:41.997502+0800 0x0        Default     0x0                  0      0    itlwm: ieee80211_vht_negotiate 804 chan_width=3 support_160=1 support_80_80=1
2021-04-24 16:50:53.575490+0800 0x0        Default     0x0                  0      0    itlwm: ieee80211_vht_negotiate 804 chan_width=3 support_160=1 support_80_80=1
2021-04-24 16:50:54.577160+0800 0x0        Default     0x0                  0      0    itlwm: iwx_rs_init line=7386

wqhqq1, 2021-04-24:
观察log发现160mhz的信道没有36（华为路由器自动选择）
我手动改成40，果然能连上160mhz

wqhqq1, 2021-04-24:
[Report_2021-04-24_16-49-27.log](https://github.com/OpenIntelWireless/itlwm/files/6369281/Report_2021-04-24_16-49-27.log)
完整log

wqhqq1, 2021-04-24:
@zxystd 我发现路由器channel设置为36的时候就是无法配速到160mhz，这是bug吗？