OpenIntelWireless/itlwm #582: 802.11ac update problems/bugs

lgariv, 2021-04-25:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: latest v2.0.0 alpha [#c448fbd](https://github.com/OpenIntelWireless/itlwm/commit/c448fbdefa681f2f59394dbb800aca2a3a50e12e)
- WiFi Card Model: AC-9462
- Product ID: `0x42A4`
- macOS Version: Catalina 10.15.7

**Description**
1. When my router is set to 80MHz only, it connects to channel 40 (only channel that can broadcast 80MHz on my router) as expected. When my router is set to 40MHz, it is also working as expected. _But_ when my router is set to 'Auto' mode (20MHz or 40MHz or 80MHz) it always detects/tries to connect over 80MHz, on channel 44 (which is incorrect), resulting in very slow speeds, unresponsiveness, and low reception. (tested only using Itlwm)
2. AirportItlwm doesn't reach speeds higher than 25mbps, even on 802.11ac 80Mhz, while Itlwm can reach 250mbps and higher. previously it was because of some limitation of Airport, due to how background scanning is managed which is more efficient on Itlwm (if I get it right?), but this kind of a difference isn't as small as it used to be.
3. Neither AirportItlwm or Itlwm are able to connect to 2.4GHz networks for me. As soon as I choose one, it get stuck on 'Connecting' until I open the menu and choose to connect to a 5GHz network (also - shouldn't there be a timeout?).

**Bug Report Archive**
[Log_2021-04-25_14-08-34.log](https://github.com/OpenIntelWireless/itlwm/files/6371780/Log_2021-04-25_14-08-34.log)

**Kext Download Source**
Releases page.

williambj1, 2021-04-25:
CC @zxystd

zxystd, 2021-04-25:
Thank you for your feedback, but I don't have time to deal with these in recent days. I'm preparing for my wedding. I'll verify and solve them when I have time in a few days. thank you.

zfdx123, 2021-04-27:
> 感谢您的反馈，但是最近几天我没有时间处理这些问题。我正在为我的婚礼做准备。几天后，我会验证并解决它们。谢谢你。

新婚快乐（Happy wedding）