OpenIntelWireless/itlwm #584: DHCP connecting problem

utteryu, 2021-04-28:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

Kext Version: latest v2.0.0 alpha #c448fbd
WiFi Card Model: ax200
Product ID: 0x2723
macOS Version: Big Sur 11.3
Description
After using the 1.3.0_stable or latest 2.0.0_alpha version, I can't get the right DHCP IP with certain router, such as OpenWRT, but work fine with other router ex, Aruba AP. 
This problem appears for both 2.4g and 5G, and setting IP manually doesn't work. The solving  way I currently use is rolling back to 1.2.0_stable, but with much lower speed.
 

**Bug Report Archive**
2.0.0 logs:
[Log_2021-04-28_14-06-59_2.0_alpha.log](https://github.com/OpenIntelWireless/itlwm/files/6389821/Log_2021-04-28_14-06-59_2.0_alpha.log)
1.3.0 logs:
[1.3.0_stable.log](https://github.com/OpenIntelWireless/itlwm/files/6389866/1.3.0_stable.log)

It always get the self-assigned IP address 169.254.xxx.xxx and will not be able to connect to the Internet.
[image]
[image]
Works fine on 1.2.0
[image]

**Kext Download Source**
Releases page.

@zxystd 
 祝新婚愉快🎉也感謝你對這個專案的努力！

xx569029747, 2021-04-28:
手动更新DHCP无效吗？

utteryu, 2021-04-28:
> 手动更新DHCP无效吗？
對，似乎根本無法和DHCP server溝通

ske1996, 2021-05-04:
I've got same problem,too

but it was a few days ago and i forgot to grab a log,sry for that.

win1010525, 2021-05-04:
CC. @zxystd 🌝

Grant07, 2021-05-05:
Having the same issue on 1.3.0, on 11.3.1.
Though I am able to bypass the problem easily by simply turning the wireless radio, on and off again, then DHCP works

tuancuong92, 2021-06-08:
Im having same issue. After renew DHCP, everything works. But sometimes, it suddenly disconnect and I cant renew DHCP again

zxystd, 2021-06-29:
So please try the latest alpha.

utteryu, 2021-06-29:
I have tried the latest alpha version #3e63230 but the issue is still there😥
[Log_2021-06-27.zip](https://github.com/OpenIntelWireless/itlwm/files/6733962/Log_2021-06-27.zip)

ske1996, 2021-06-30:
> So please try the latest alpha.

dhcp works great in lastest version,thank you so much

zxystd, 2021-07-24:
Thank you for the feedback.

closed 2021-07-24 by zxystd

MaxSYC, 2021-07-26:
I have tested the  #fc497c1 alpha, and all works properly. If any logs needed please let me know, many thanks for your help.