OpenIntelWireless/itlwm #585: 笔记本（8265网卡），睡眠后wifi经常断流并自动重连数次

zxbu, 2021-05-04:
笔记本型号：Lg gram 14z970
网卡：8265
驱动：1.3.0
版本：mac 10.15.7 + opencore 0.6.8

现象：
笔记本从睡眠唤醒后，大概率会每几分钟就丢包并自动重连数次。即使关闭重打开WiFi开关也没用，重启电脑一般可以解决问题。
但只要睡眠过，又会出现此现象

[Log_2021-05-04_08-46-39.log](https://github.com/OpenIntelWireless/itlwm/files/6418566/Log_2021-05-04_08-46-39.log)

gongyc, 2021-05-04:
一样的问题

highf4321, 2021-05-06:
Same Issue on 9560AC.

zxystd, 2021-05-07:
@zxbu  The log is incomplete, please collect another full one.
hmmm, I don't think the abnormally symptom is caused by the driver, it is mostly like a Hackintosh issue, you should fix it by your self.

> Same Issue on 9560AC.

@gongyc @highf4321  If you have same issue, please provide the logs.

zxbu, 2021-05-08:
@zxystd 有可能吧，我对黑苹果不太擅长，我没其他问题了。
你可以把这个issue关闭

zxystd, 2021-05-09:
Are you connecting a mesh type of WIFI?

zxbu, 2021-05-10:
> Are you connecting a mesh type of WIFI?

对，型号是 水星AX1800，共两个，通过有线组网，wifi名字双频合一

fsnhljz, 2021-06-02:
一样的问题，我不认为是黑苹果引起的。换用外置网卡没这样的问题。

zxbu, 2021-08-12:
我升级到2.0.0 稳定版后再无出现过此类问题，我先关了问题哈

closed 2021-08-12 by zxbu