OpenIntelWireless/itlwm #588: 9260AC WIFI is not working.

WarsFeng, 2021-05-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v1.3.0 stable / v1.2.0 stable
- WiFi Card Model: Wireless-AC 9260
- Product ID:  Vendor: **0x8086**, Device: **0x2526**
- macOS Version: BigSur 11.2.3

**Description**

9260AC WIFI is not working.

**Bug Report Archive**
[bugreport_40165.zip](https://github.com/OpenIntelWireless/itlwm/files/6445249/bugreport_40165.zip)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.3.0

zxystd, 2021-05-09:
I am also using AC9260 for testing and it works fine.
can you help me to use this debug kext and then generate the log again? I need more  debug info, thanks. @WarsFeng 
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6447647/itlwm.kext.zip)

WarsFeng, 2021-05-11:
This is the new kext log:
[bugreport_55517.zip](https://github.com/OpenIntelWireless/itlwm/files/6456048/bugreport_55517.zip)

bluetooth is working properly.

zxystd, 2021-05-11:
@WarsFeng Have you tried linux, is this card work properly?

WarsFeng, 2021-05-11:
Oh my god, linux-firmware(Ubuntu/Arch livecd) not work... not show WiFi device, but Windows works.

zxystd, 2021-05-11:
@WarsFeng seems that your card didn't have REGULATORY section info, which is help to decide channel/country code. I think this is a bug about Linux iwlwifi or a hardware bug.
[image]
I think you can see this error message in linux.

zxystd, 2021-05-11:
@WarsFeng I have mod something, can you see if it can load?
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6456767/itlwm.kext.zip)

zxystd, 2021-05-12:
@WarsFeng Hello?

WarsFeng, 2021-05-13:
@zxystd It doesn't seem to load =.=, this is the new log:
[bugreport_55517.zip](https://github.com/OpenIntelWireless/itlwm/files/6471279/bugreport_55517.zip)

Want to give up and change ax200 QAQ.

WarsFeng, 2021-05-13:
The manual load log is as follows:
```
Executing: /usr/bin/kmutil load -p /Users/wars/Downloads/itlwm.kext
Error Domain=KMErrorDomain Code=30 "Validating extension failed: Kext com.zxystd.itlwm v2.0.0 in executable kext bundle com.zxystd.itlwm at /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext:

Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents/MacOS
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents/MacOS/itlwm
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents/Info.plist
Authenticating extension failed: Bad code signature" UserInfo={NSLocalizedDescription=Validating extension failed: Kext com.zxystd.itlwm v2.0.0 in executable kext bundle com.zxystd.itlwm at /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext:

Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents/MacOS
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents/MacOS/itlwm
Filesystem error: Invalid ownership (501:20) should be (0:0) on /private/var/db/KernelExtensionManagement/Staging/com.zxystd.itlwm.7LScJi/itlwm.kext/Contents/Info.plist
Authenticating extension failed: Bad code signature}
```

closed 2021-05-19 by WarsFeng