OpenIntelWireless/itlwm #590: Version 2.0 refuses to connect to any router Wireless-AC 9560

theroadw, 2021-05-12:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: itlwm 2.0
- WiFi Card Model:  Wireless-AC 9560
- Product ID: A370
- macOS Version: macOS 10.14.6 (18G103)

**Description**

Version 2.0 refuses to connect to any router, I included the diagnostic report using the unmodified latest commit version.

Now in order to make Version 2.0 connect, I used the latest commit, but reverted the changes done here: https://github.com/OpenIntelWireless/itlwm/commit/c448fbdefa681f2f59394dbb800aca2a3a50e12e

with these changes wifi will connect, but speed is not AC, even if Heliport claims connection is @867Mbps

The link speed as detected by the router is @300Mbps and iperf tests pretty much show the same.

My network has a Gigabit backend and you can see the Ethernet iperf test as baseline.

Using Linux, Wifi-AC shows ~ 80MBps. Windows is about the same.

* Version 1.3 Stable works perfectly, Bluetooth also.

Thank you for your work on this

**Bug Report Archive**

[bugreport_46113.zip](https://github.com/OpenIntelWireless/itlwm/files/6468460/bugreport_46113.zip)

**Kext Download Source**

Custom built

zxystd, 2021-05-20:
hmmm, nothing change with that commit... have you tried reseting the network?

theroadw, 2021-05-20:
Yes I tried resetting the network and it made no difference. I also don't see anything in the code that could have created this problem, but before this code, the connection problem wasn't there, and after it no connections could be established, and using the latest version but regressing only those changes, I can establish connections again.

m1140, 2021-05-24:
With new VHT capacity, it looks like the driver has problem receiving packets. Not sure if this is related to OS, but I'm also running Mojave 10.14.6...

theroadw, 2021-05-28:
This commit seems to have fixed the connection problem:
https://github.com/OpenIntelWireless/itlwm/commit/604a33b0d8fb7fa0025ff2412036cbf90e030489

Only weird thing in the router I can see connection speed drops when executing iperf, and as soon as it finishes, it comes back up.
The test results show an improvement from N, and are 1/2 the transfer speed from the linux/windows drivers.

[image]

[image]

closed 2021-05-28 by theroadw