OpenIntelWireless/itlwm #594: itlwm v2.0 crash with ax201 in bigsur

MoeGuoH, 2021-05-17:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: v2.0.0-alpha 
- WiFi Card Model: Intel AX201
- Product ID: 
- macOS Version: 11.3.1

**Description**

random time crash.

**Bug Report Archive**

```
panic(cpu 2 caller 0xffffff801c9e6d8e): sta already removed
Backtrace (CPU 2), Frame : Return Address
0xffffffa0bbb939e0 : 0xffffff801888e02d mach_kernel : _handle_debugger_trap + 0x3fd
0xffffffa0bbb93a30 : 0xffffff80189d48e3 mach_kernel : _kdp_i386_trap + 0x143
0xffffffa0bbb93a70 : 0xffffff80189c4eda mach_kernel : _kernel_trap + 0x55a
0xffffffa0bbb93ac0 : 0xffffff8018832a2f mach_kernel : _return_from_trap + 0xff
0xffffffa0bbb93ae0 : 0xffffff801888d84d mach_kernel : _DebuggerTrapWithState + 0xad
0xffffffa0bbb93c00 : 0xffffff801888db43 mach_kernel : _panic_trap_to_debugger + 0x273
0xffffffa0bbb93c70 : 0xffffff801909d68a mach_kernel : _panic + 0x54
0xffffffa0bbb93ce0 : 0xffffff801c9e6d8e com.zxystd.AirportItlwm : __ZN6ItlIwx14iwx_rm_sta_cmdEP9iwx_softcP8iwx_node + 0x4e
0xffffffa0bbb93d30 : 0xffffff801c9eaa94 com.zxystd.AirportItlwm : __ZN6ItlIwx12iwx_disassocEP9iwx_softc + 0x174
0xffffffa0bbb93dd0 : 0xffffff801c9ec0d9 com.zxystd.AirportItlwm : __ZN6ItlIwx17iwx_newstate_taskEPv + 0x1b9
0xffffffa0bbb93e40 : 0xffffff801c97a73f com.zxystd.AirportItlwm : __Z12taskq_threadPv + 0x6f
0xffffffa0bbb93fa0 : 0xffffff801883213e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.zxystd.AirportItlwm(2.0)[FD64807D-B468-39FD-83E5-B63808E30F6A]@0xffffff801c977000->0xffffff801d6d1fff
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[850659DB-5F03-349E-8923-ADDF3C7E0DF0]@0xffffff801a309000->0xffffff801a488fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[EFF68ECC-483B-3BAC-82B9-09C8F0600009]@0xffffff801b0d2000->0xffffff801b0e8fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[B99B3B05-71B8-3178-8C27-E0A0ACA9F16E]@0xffffff801b35f000->0xffffff801b387fff

Process name corresponding to current thread: kernel_task
Boot args: keepsyms=1 brcmfx-country=#a chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
20E241

Kernel version:
Darwin Kernel Version 20.4.0: Thu Apr 22 21:46:47 PDT 2021; root:xnu-7195.101.2~1/RELEASE_X86_64
Kernel UUID: 068F99A3-1DB3-31C0-87D5-09942F122BB6
KernelCache slide: 0x0000000018600000
KernelCache base:  0xffffff8018800000
Kernel slide:      0x0000000018610000
Kernel text base:  0xffffff8018810000
__HIB  text base: 0xffffff8018700000
System model name: MacBookPro16,2 (Mac-5F9802EFE386AA28)
System shutdown begun: NO
Panic diags file available: YES (0x0)
Hibernation exit count: 0

System uptime in nanoseconds: 8079282358803
Last Sleep:           absolute           base_tsc          base_nano
  Uptime  : 0x000007591ac15105
  Sleep   : 0x0000058e6d4f18b8 0x00000d823fc6b028 0x000004075fa7adea
  Wake    : 0x0000058e6f0e42d3 0x000010cb913dfcf2 0x0000058e6e1e2665
last started kext at 1934778154725: com.parallels.kext.usbconnect	16.1.2 49151 (addr 0xffffff7fb39aa000, size 20480)
last stopped kext at 214476059528: >IOPlatformPluginLegacy	1.0.0 (addr 0xffffff7fb1f04000, size 36864)
loaded kexts:
com.parallels.kext.usbconnect	16.1.2 49151
com.parallels.kext.netbridge	16.1.2 49151
com.parallels.kext.vnic	16.1.2 49151
com.parallels.kext.hypervisor	16.1.2 49151
com.intel.driver.EnergyDriver	3.7.0
me.kishorprins.VoodooInput	1.1.3
com.alexandred.VoodooI2CHID	1
as.acidanthera.voodoo.driver.PS2Keyboard	2.2.4
com.zxystd.!I!BFirmware	1.1.2
com.zxystd.AirportItlwm	2.0.0
com.alexandred.VoodooI2C	2.6.5
org.coolstar.VoodooGPIO	1.1
as.acidanthera.voodoo.driver.PS2!C	2.2.4
as.vit9696.SMCProcessor	1.2.4
as.acidanthera.BrightnessKeys	1.0.1
com.alexandred.VoodooI2CServices	1
as.vit9696.!AALC	1.6.1
ru.usrsse2.SMCLightSensor	1.2.4
ru.usrsse2.SMCBatteryManager	1.2.4
as.vit9696.VirtualSMC	1.2.4
as.vit9696.WhateverGreen	1.5.0
as.vit9696.Lilu	1.5.4
@nke.l2tp	1.9
>AudioAUUC	1.70
>!APlatformEnabler	2.7.0d0
>AGPM	122
>X86PlatformShim	1.0.0
@filesystems.autofs	3.0
@fileutil	20.036.15
>!AHDA	283.15
>!AGraphicsDevicePolicy	6.2.9
>!AUpstreamUserClient	3.6.8
@AGDCPluginDisplayMetrics	6.2.9
>pmtelemetry	1
|IOUserEthernet	1.0.1
>usb.!UUserHCI	1
|IO!BSerialManager	8.0.4d18
@Dont_Steal_Mac_OS_X	7.0.0
>!A!IKBLGraphics	16.0.2
>!AHV	1
>!ADiskImages2	1
>!ASMCLMU	212
>!A!IPCHPMC	2.0.1
>!A!ISlowAdaptiveClocking	4.0.0
>!A!ICFLGraphicsFramebuffer	16.0.2
>!ABacklight	180.3
>!AMCCSControl	1.14
>!ATopCaseHIDEventDriver	4040.11
>!AFileSystemDriver	3.0.1
@filesystems.tmpfs	1
@filesystems.hfs.kext	556.100.11
@BootCache	40
@!AFSCompression.!AFSCompressionTypeZlib	1.0.0
@!AFSCompression.!AFSCompressionTypeDataless	1.0.0d1
>!ASmartBatteryManager	161.0.0
@private.KextAudit	1.0
@filesystems.apfs	1677.100.114
>!ARTC	2.0
>!AHPET	1.8
>!AACPIButtons	6.1
>!ASMBIOS	2.1
>!AACPIEC	6.1
>!AAPIC	1.7
@!ASystemPolicy	2.0.0
@nke.applicationfirewall	311
|IOKitRegistryCompatibility	1
|EndpointSecurity	1
@nke.ppp	1.9
@kext.triggers	1.0
>DspFuncLib	283.15
@kext.OSvKernDSPLib	529
>!AGraphicsControl	6.2.9
|IOAVB!F	940.4
|IOEthernetAVB!C	1.1.0
@!AGPUWrangler	6.2.9
|IOSlowAdaptiveClocking!F	1.0.0
|IOAccelerator!F2	442.9
>!ASMBusPCI	1.0.14d1
|IOSerial!F	11
@!AGraphicsDeviceControl	6.2.9
>!AHDA!C	283.15
|IOHDA!F	283.15
|IOAudio!F	300.6.1
@vecLib.kext	1.2.0
@plugin.IOgPTPPlugin	980.4
>!ABacklightExpert	1.1.0
|IONDRVSupport	585.1
|IOGraphics!F	585.1
>!ASMBus!C	1.0.18d1
>X86PlatformPlugin	1.0.0
>IOPlatformPlugin!F	6.0.0d8
>!AActuatorDriver	4440.3
>!AMultitouchDriver	4440.3
>!AInputDeviceSupport	4400.35
>!AHS!BDriver	4040.11
>IO!BHIDDriver	8.0.4d18
>usb.cdc	5.0.0
>usb.networking	5.0.0
>usb.!UHostCompositeDevice	1.2
|Broadcom!BHost!CUSBTransport	8.0.4d18
|IO!BHost!CUSBTransport	8.0.4d18
|IO!BHost!CTransport	8.0.4d18
>!UMergeNub	900.4.2
>!ABSDKextStarter	3
|IOSurface	290.7
@filesystems.hfs.encodings.kext	1
>usb.!UHostPacketFilter	1.0
|IOUSB!F	900.4.2
>!AXsanScheme	3
|IONVMe!F	2.1.0
>!AEFINVRAM	2.1
|IO80211!F	1200.12.2b1
|IOSkywalk!F	1
>mDNSOffloadUserClient	1.0.1b8
>corecapture	1.0.4
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
>!AEFIRuntime	2.1
$!AImage4	3.0.0
|IOHID!F	2.0.0
|IOSMBus!F	1.1
|IOTimeSync!F	980.4
|IONetworking!F	3.4
>DiskImages	493.0.0
|IO!B!F	8.0.4d18
|IOReport!F	47
|IO!BPacketLogger	8.0.4d18
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
|CoreAnalytics!F	1
>!ASSE	1.0
>!AKeyStore	2
>!UTDM	511.101.1
|IOUSBMass!SDriver	184.101.1
|IOSCSIBlockCommandsDevice	436.100.4
|IO!S!F	2.1
|IOSCSIArchitectureModel!F	436.100.4
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!ACredentialManager	1.0
>KernelRelayHost	1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
>!AACPIPlatform	6.1
>!ASMC	3.1.9
|IOPCI!F	2.9
|IOACPI!F	1.4
>watchdog	1
@kec.pthread	1
@kec.corecrypto	11.1
@kec.Libm	1
```

**Kext Download Source**

zxystd, 2021-05-17:
I don't think it is 'random', it should be somewhere racing, how do you reproduce it?

MoeGuoH, 2021-05-17:
> I don't think it is 'random', it should be somewhere racing, how do you reproduce it?

ah,emmm,I met twice. once i just browse the website use chrome ,sec is writing code use vscode,and than system all crash without warning ,when i after reboot system i get this system bug report ,it may be an accident,i don't have idea about this crash.So I want to send this bug rep out to see if anyone has the same experience

closed 2021-05-23 by MoeGuoH