OpenIntelWireless/itlwm #596: A crash log from the newest Airportitlwm version

kingo132, 2021-05-18:
- Kext Version: 2.0.0
- WiFi Card Model: AX200
- Product ID: 8086:2723
- macOS Version: 10.15.7

**Description**
System crash when idle, I left the device open before sleep, the next day morning, the device is dead. After reboot I found this log.

```
panic(cpu 2 caller 0xffffff8000e43a5a): Kernel trap at 0xffffff8000b980ba, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x00000000070a8000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff802305bb40, RCX: 0x0000000000000100, RDX: 0x0000000000000100
RSP: 0xffffff81ef7d3a38, RBP: 0xffffff81ef7d3d10, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000080, R10: 0x0000000000000008, R11: 0x0000000000000000
R12: 0x00000000e00002cc, R13: 0xffffff81ec07d090, R14: 0xffffff7f84c79740, R15: 0xffffff802324b230
RFL: 0x0000000000010246, RIP: 0xffffff8000b980ba, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000002, Fault CPU: 0x2, PL: 0, VF: 1

Backtrace (CPU 2), Frame : Return Address
0xffffff81ef7d3390 : 0xffffff8000d1763d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff81ef7d33e0 : 0xffffff8000e51b25 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81ef7d3420 : 0xffffff8000e436ae mach_kernel : _kernel_trap + 0x4ee
0xffffff81ef7d3470 : 0xffffff7f8497a924 as.vit9696.VirtualSMC : __ZN18VirtualSMCProvider10kernelTrapI22x86_saved_state_1010_tEEvPT_Pm + 0x454
0xffffff81ef7d34f0 : 0xffffff8000cbda40 mach_kernel : _return_from_trap + 0xe0
0xffffff81ef7d3510 : 0xffffff8000d16d07 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff81ef7d3610 : 0xffffff8000d170f7 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff81ef7d3660 : 0xffffff80014be82c mach_kernel : _panic + 0x54
0xffffff81ef7d36d0 : 0xffffff8000e43a5a mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff81ef7d3850 : 0xffffff8000e43758 mach_kernel : _kernel_trap + 0x598
0xffffff81ef7d38a0 : 0xffffff7f8497a924 as.vit9696.VirtualSMC : __ZN18VirtualSMCProvider10kernelTrapI22x86_saved_state_1010_tEEvPT_Pm + 0x454
0xffffff81ef7d3920 : 0xffffff8000cbda40 mach_kernel : _return_from_trap + 0xe0
0xffffff81ef7d3940 : 0xffffff8000b980ba 
0xffffff81ef7d3d10 : 0xffffff7f84c7993d com.zxystd.AirportItlwm : __ZN6ItlIwx15_iwx_start_taskEP8OSObjectPvS2_S2_S2_ + 0x1fd
0xffffff81ef7d3da0 : 0xffffff800142b9e5 mach_kernel : __ZN13IOCommandGate13attemptActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb5
0xffffff81ef7d3e00 : 0xffffff7f84c79ad2 com.zxystd.AirportItlwm : __ZN6ItlIwx9iwx_startEP6_ifnet + 0x92
0xffffff81ef7d3e50 : 0xffffff7f84cbbd39 com.zxystd.AirportItlwm : __ZN12AirportItlwm12outputPacketEP6__mbufPv + 0x219
0xffffff81ef7d3eb0 : 0xffffff7f84cbbab5 com.zxystd.AirportItlwm : __ZN12AirportItlwm11outputStartEP18IONetworkInterfacej + 0xc5
0xffffff81ef7d3f20 : 0xffffff7f81abed04 com.apple.iokit.IONetworkingFamily : __ZN18IONetworkInterface16drainOutputQueueEP7__ifnetP19IONetworkController + 0x5a
0xffffff81ef7d3f60 : 0xffffff8000ff3af5 mach_kernel : _ifnet_attach + 0x25a5
0xffffff81ef7d3fa0 : 0xffffff8000cbd13e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[D45CAA28-BC73-3E06-BBCA-48E5D17D764B]@0xffffff7f81ab0000->0xffffff7f81adffff
         as.vit9696.VirtualSMC(1.1.9)[3B5840A3-EFD8-3F34-B9DA-9DDB414549B3]@0xffffff7f8496b000->0xffffff7f84991fff
            dependency: as.vit9696.Lilu(1.5.2)[37C5AEFB-3280-360A-887A-62A05F24BCF6]@0xffffff7f848e4000
            dependency: com.apple.iokit.IOACPIFamily(1.4)[0BC116A3-534A-3990-99CB-09DB34A5EFAA]@0xffffff7f81709000
         com.zxystd.AirportItlwm(2.0)[D7EC5934-B71B-3C34-BA38-303DB46C983D]@0xffffff7f84c04000->0xffffff7f85958fff
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[16B44F36-3630-373E-BB78-BB131863B1CB]@0xffffff7f8266c000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[D45CAA28-BC73-3E06-BBCA-48E5D17D764B]@0xffffff7f81ab0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[B4389948-0918-363E-8E70-0FBC89B384CA]@0xffffff7f81712000

BSD process name corresponding to current thread: kernel_task
Boot args: -v -igfxcdc -igfxdvmt keepsyms=1 debug=0x100 darkwake=0 dc6config=0 agdpmod=pikera msgbuf=1048576 igfxfw=2 chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
19H1030

Kernel version:
Darwin Kernel Version 19.6.0: Mon Apr 12 20:57:45 PDT 2021; root:xnu-6153.141.28.1~1/RELEASE_X86_64
Kernel UUID: BB1171FD-A9E3-3D24-92F7-F4DD45974EFD
Kernel slide:     0x0000000000a00000
Kernel text base: 0xffffff8000c00000
__HIB  text base: 0xffffff8000b00000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 24979647086604

```

Andy-math, 2021-05-19:
#### Very similar backtrace. 

- Kext Version: 1.3.0 stable
- WiFi Card Model: AX201
- Product ID: 8086:02ed
- macOS Version: 10.15.7

#### Description
System crash when idle, might happen when vscode is running. There is usually abnormal CPU usage prior to this, in the form of fan noise.
```
panic(cpu 6 caller 0xffffff8004e43a5a): Kernel trap at 0xffffff8004b980ba, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000001caf5000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff8038ca3e10, RCX: 0x0000000000000100, RDX: 0x0000000000000100
RSP: 0xffffff921d133a58, RBP: 0xffffff921d133d10, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000080, R10: 0x0000000000000008, R11: 0x0000000000000000
R12: 0x00000000e00002cc, R13: 0xffffff81f7eaa090, R14: 0xffffff7f88c52480, R15: 0xffffff8038d81870
RFL: 0x0000000000010246, RIP: 0xffffff8004b980ba, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000002, Fault CPU: 0x6, PL: 0, VF: 1

Backtrace (CPU 6), Frame : Return Address
0xffffff921d1334b0 : 0xffffff8004d1763d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff921d133500 : 0xffffff8004e51b25 mach_kernel : _kdp_i386_trap + 0x155
0xffffff921d133540 : 0xffffff8004e436ae mach_kernel : _kernel_trap + 0x4ee
0xffffff921d133590 : 0xffffff8004cbda40 mach_kernel : _return_from_trap + 0xe0
0xffffff921d1335b0 : 0xffffff8004d16d07 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff921d1336b0 : 0xffffff8004d170f7 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff921d133700 : 0xffffff80054be82c mach_kernel : _panic + 0x54
0xffffff921d133770 : 0xffffff8004e43a5a mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff921d1338f0 : 0xffffff8004e43758 mach_kernel : _kernel_trap + 0x598
0xffffff921d133940 : 0xffffff8004cbda40 mach_kernel : _return_from_trap + 0xe0
0xffffff921d133960 : 0xffffff8004b980ba 
0xffffff921d133d10 : 0xffffff7f88c5268e com.zxystd.itlwm : __ZN6ItlIwx15_iwx_start_taskEP8OSObjectPvS2_S2_S2_ + 0x20e
0xffffff921d133da0 : 0xffffff800542b9e5 mach_kernel : __ZN13IOCommandGate13attemptActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb5
0xffffff921d133e00 : 0xffffff7f88c52822 com.zxystd.itlwm : __ZN6ItlIwx9iwx_startEP6_ifnet + 0x92
0xffffff921d133e50 : 0xffffff7f88c8b389 com.zxystd.itlwm : __ZN5itlwm12outputPacketEP6__mbufPv + 0x219
0xffffff921d133eb0 : 0xffffff7f88c8b105 com.zxystd.itlwm : __ZN5itlwm11outputStartEP18IONetworkInterfacej + 0xc5
0xffffff921d133f20 : 0xffffff7f85abed04 com.apple.iokit.IONetworkingFamily : __ZN18IONetworkInterface16drainOutputQueueEP7__ifnetP19IONetworkController + 0x5a
0xffffff921d133f60 : 0xffffff8004ff3af5 mach_kernel : _ifnet_attach + 0x25a5
0xffffff921d133fa0 : 0xffffff8004cbd13e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[D45CAA28-BC73-3E06-BBCA-48E5D17D764B]@0xffffff7f85ab0000->0xffffff7f85adffff
         com.zxystd.itlwm(1.3)[1EB4A67B-0FED-3F45-A1DD-B7DA5FBC5208]@0xffffff7f88c3a000->0xffffff7f89975fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[D45CAA28-BC73-3E06-BBCA-48E5D17D764B]@0xffffff7f85ab0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[B4389948-0918-363E-8E70-0FBC89B384CA]@0xffffff7f85712000

BSD process name corresponding to current thread: kernel_task
Boot args: -v keepsyms=1 alcid=11 brcmfx-country=#a chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
19H1030

Kernel version:
Darwin Kernel Version 19.6.0: Mon Apr 12 20:57:45 PDT 2021; root:xnu-6153.141.28.1~1\/RELEASE_X86_64
Kernel UUID: BB1171FD-A9E3-3D24-92F7-F4DD45974EFD
Kernel slide:     0x0000000004a00000
Kernel text base: 0xffffff8004c00000
__HIB  text base: 0xffffff8004b00000
System model name: MacBookPro16,2 (Mac-5F9802EFE386AA28)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 144662264553302
last loaded kext at 137990434751488: >!AHDAHardwareConfigDriver	283.15 (addr 0xffffff7f8a919000, size 12288)
last unloaded kext at 131025417579227: com.parallels.kext.hypervisor	14.1.3 45485 (addr 0xffffff7f8b253000, size 204800)
loaded kexts:
com.intel.driver.EnergyDriver	3.7.0
com.zxystd.!I!BFirmware	1.1.2
me.kishorprins.VoodooInput	1.1.1
com.alexandred.VoodooI2CHID	1
as.acidanthera.voodoo.driver.PS2Keyboard	2.2.3
[... 125 lines cut ...]
|IONetworking!F	3.4
>DiskImages	493.0.0
|IO!B!F	7.0.6f8
|IO!BPacketLogger	7.0.6f8
>!ASSE	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.140.1
|IOSCSIArchitectureModel!F	422.120.3
|IO!S!F	2.1
|IOUSBHost!F	1.2
>usb.!UCommon	1.0
>!UHostMergeProperties	1.2
>!ABusPower!C	1.0
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1
```

zxystd, 2021-05-20:
@kingo132 @Andy-math would you mind uploading the itlwm kext you are using? I can't match the git version.

Andy-math, 2021-05-20:
@zxystd The itlwm130 I am using is from ` https://github.com/OpenIntelWireless/itlwm/releases/download/v1.3.0/itlwm_v1.3.0_stable.kext.zip ` , if the URL has not changed since Sunday, March 16th (UTC+8).

Sincerely thank you for your continued work on this project.

zxystd, 2021-05-20:
@Andy-math I download the 1.3.0 stable kext, but can't match the crash symbol, seems that you are not using this version.

Andy-math, 2021-05-20:
@zxystd Sorry about that
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6512710/itlwm.kext.zip)

Andy-math, 2021-05-20:
I used Beyond Compare to compare the kext in my efi with the stable release that the URL points to, and I couldn't find any difference. Maybe there's a problem with my configuration elsewhere? Does OpenCore need `touch s/l/e`?
```
$ kextstat | grep -i "itlwm"
   72    0 0xffffff7f8423a000 0xd3c000   0xd3c000   com.zxystd.itlwm (1.3.0) 1EB4A67B-0FED-3F45-A1DD-B7DA5FBC5208 <32 13 6 5 3 1>
```

Andy-math, 2021-05-20:
After `touch s/l/e` and reboot, it's same: 
```
$ kextstat | grep -i "itlwm"
   72    0 0xffffff7f8423a000 0xd3c000   0xd3c000   com.zxystd.itlwm (1.3.0) 1EB4A67B-0FED-3F45-A1DD-B7DA5FBC5208 <32 13 6 5 3 1>
```
But I still can't find `itlwm` from `About This Mac` > `System Report`, maybe I'm going the wrong way

Andy-math, 2021-05-20:
`$ nm -g /Volumes/Untitled/EFI/OC/Kexts/itlwm.kext/Contents/MacOS/itlwm > issue596_symbol130.txt`
[issue596_symbol130.txt](https://github.com/OpenIntelWireless/itlwm/files/6512904/issue596_symbol130.txt)

kingo132, 2021-05-20:
> @Andy-math I download the 1.3.0 stable kext, but can't match the crash symbol, seems that you are not using this version.

Here is the one I'm using

[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6513478/AirportItlwm.kext.zip)

Andy-math, 2021-05-25:
again
'''
panic(cpu 0 caller 0xffffff8005e43a5a): Kernel trap at 0xffffff8005b980ba, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000001daf5000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff8039bc1c30, RCX: 0x0000000000000100, RDX: 0x0000000000000100
RSP: 0xffffff92487b3a58, RBP: 0xffffff92487b3d10, RSI: 0x0000000000000000, RDI: 0x0000000000000000
R8:  0x0000000000000000, R9:  0x0000000000000080, R10: 0x0000000000000008, R11: 0x0000000000000000
R12: 0x00000000e00002cc, R13: 0xffffff81f8e6d090, R14: 0xffffff7f89c52480, R15: 0xffffff8039cc5be0
RFL: 0x0000000000010246, RIP: 0xffffff8005b980ba, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000002, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff92487b34b0 : 0xffffff8005d1763d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff92487b3500 : 0xffffff8005e51b25 mach_kernel : _kdp_i386_trap + 0x155
0xffffff92487b3540 : 0xffffff8005e436ae mach_kernel : _kernel_trap + 0x4ee
0xffffff92487b3590 : 0xffffff8005cbda40 mach_kernel : _return_from_trap + 0xe0
0xffffff92487b35b0 : 0xffffff8005d16d07 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff92487b36b0 : 0xffffff8005d170f7 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff92487b3700 : 0xffffff80064be82c mach_kernel : _panic + 0x54
0xffffff92487b3770 : 0xffffff8005e43a5a mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff92487b38f0 : 0xffffff8005e43758 mach_kernel : _kernel_trap + 0x598
0xffffff92487b3940 : 0xffffff8005cbda40 mach_kernel : _return_from_trap + 0xe0
0xffffff92487b3960 : 0xffffff8005b980ba 
0xffffff92487b3d10 : 0xffffff7f89c5268e com.zxystd.itlwm : __ZN6ItlIwx15_iwx_start_taskEP8OSObjectPvS2_S2_S2_ + 0x20e
0xffffff92487b3da0 : 0xffffff800642b9e5 mach_kernel : __ZN13IOCommandGate13attemptActionEPFiP8OSObjectPvS2_S2_S2_ES2_S2_S2_S2_ + 0xb5
0xffffff92487b3e00 : 0xffffff7f89c52822 com.zxystd.itlwm : __ZN6ItlIwx9iwx_startEP6_ifnet + 0x92
0xffffff92487b3e50 : 0xffffff7f89c8b389 com.zxystd.itlwm : __ZN5itlwm12outputPacketEP6__mbufPv + 0x219
0xffffff92487b3eb0 : 0xffffff7f89c8b105 com.zxystd.itlwm : __ZN5itlwm11outputStartEP18IONetworkInterfacej + 0xc5
0xffffff92487b3f20 : 0xffffff7f86abed04 com.apple.iokit.IONetworkingFamily : __ZN18IONetworkInterface16drainOutputQueueEP7__ifnetP19IONetworkController + 0x5a
0xffffff92487b3f60 : 0xffffff8005ff3af5 mach_kernel : _ifnet_attach + 0x25a5
0xffffff92487b3fa0 : 0xffffff8005cbd13e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IONetworkingFamily(3.4)[D45CAA28-BC73-3E06-BBCA-48E5D17D764B]@0xffffff7f86ab0000->0xffffff7f86adffff
         com.zxystd.itlwm(1.3)[1EB4A67B-0FED-3F45-A1DD-B7DA5FBC5208]@0xffffff7f89c3a000->0xffffff7f8a975fff
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[D45CAA28-BC73-3E06-BBCA-48E5D17D764B]@0xffffff7f86ab0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[B4389948-0918-363E-8E70-0FBC89B384CA]@0xffffff7f86712000

BSD process name corresponding to current thread: kernel_task
Boot args: -v keepsyms=1 alcid=11 brcmfx-country=#a chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
19H1030

Kernel version:
Darwin Kernel Version 19.6.0: Mon Apr 12 20:57:45 PDT 2021; root:xnu-6153.141.28.1~1/RELEASE_X86_64
Kernel UUID: BB1171FD-A9E3-3D24-92F7-F4DD45974EFD
Kernel slide:     0x0000000005a00000
Kernel text base: 0xffffff8005c00000
__HIB  text base: 0xffffff8005b00000
System model name: MacBookPro16,2 (Mac-5F9802EFE386AA28)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 378839684391921
last loaded kext at 262149719203123: @filesystems.smbfs	3.4.4 (addr 0xffffff7f8c30d000, size 450560)
loaded kexts:
com.parallels.kext.netbridge	14.1.3 45485
com.parallels.kext.hypervisor	14.1.3 45485
com.parallels.kext.vnic	14.1.3 45485
com.intel.driver.EnergyDriver	3.7.0
com.zxystd.!I!BFirmware	1.1.2
me.kishorprins.VoodooInput	1.1.1
as.acidanthera.voodoo.driver.PS2Keyboard	2.2.3
com.alexandred.VoodooI2CHID	1
com.alexandred.VoodooI2C	2.6.5
com.zxystd.itlwm	1.3.0
as.acidanthera.voodoo.driver.PS2!C	2.2.3
org.coolstar.VoodooGPIO	1.1
as.vit9696.SMCProcessor	1.2.2
as.acidanthera.BrightnessKeys	1.0.1
[... 125 lines cut ...]
|IONetworking!F	3.4
>DiskImages	493.0.0
|IO!B!F	7.0.6f8
|IO!BPacketLogger	7.0.6f8
>!ASSE	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOUSBMass!SDriver	157.140.1
|IOSCSIArchitectureModel!F	422.120.3
|IO!S!F	2.1
|IOUSBHost!F	1.2
>usb.!UCommon	1.0
>!UHostMergeProperties	1.2
>!ABusPower!C	1.0
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.corecrypto	1.0
@kec.Libm	1
'''

zxystd, 2021-05-30:
hmm, I don't have any clue on this, maybe it is a sleep issue? what do you mean "idle" state? @Andy-math @kingo132

Andy-math, 2021-05-30:
@zxystd I'm not sure. My condition occurred shortly after I used the ```Trigger Angle``` - ```put the monitor to sleep```. Otherwise, after successfully sleeping for a period of time, there is no longer any possibility of panic until I wake it up again and put it back to sleep.
Just putting the monitor to sleep means keeping it responsive to the USB (it doesn't seem to have completely suspended the wireless card), which is different from sleeping with the lid closed after the power is disconnected.
Close the lid after disconnecting the power source is almost never tried.

kingo132, 2021-05-31:
> hmm, I don't have any clue on this, maybe it is a sleep issue? what do you mean "idle" state? @Andy-math @kingo132

Idle means I do not touch the computer for a night, first, the display will go to sleep, then the whole system will sleep too, and the next day morning I found the computer is unresponsive. I couldn't figure out when the crash happens. It is between I left the computer alone and the next day morning.

williambj1, 2021-12-26:
Hi @kingo132 @Andy-math 

Does the panic still appear when using the latest RC? Thanks!

williambj1, 2022-03-03:
Feel free to add comments if the same panic still appears.

closed 2022-03-03 by williambj1