OpenIntelWireless/itlwm #597: Airportitlwm often lost wifi connection and then connect again within 3 seconds

kingo132, 2021-05-18:
Kext Version: 2.0.0
WiFi Card Model: AX200
Product ID: 8086:2723
macOS Version: 10.15.7

**Description**
Here is the log when it lost connection and reconnect again. Hope this will help
```
jason@mywm ~ % log show --start '2021-5-18 08:58:00' --end '2021-5-18 09:00:00' | grep -i itlwm
2021-05-18 08:58:03.531600+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_tx type=8 qos=0 multicast=0 len=24 subtype=0 qid=65535 using mgmt tid
2021-05-18 08:58:05.921621+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_tx type=8 qos=0 multicast=0 len=24 subtype=0 qid=65535 using mgmt tid
2021-05-18 08:58:10.512683+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_rs_update new rate: 0x4c00f018: VHT | ANT: AB BW: 80Mhz MCS: 8 NSS: 2 SGI LDPC
2021-05-18 08:58:15.931140+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_tx type=8 qos=0 multicast=0 len=24 subtype=0 qid=65535 using mgmt tid
2021-05-18 08:58:23.148554+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_rs_update new rate: 0x4c00f017: VHT | ANT: AB BW: 80Mhz MCS: 7 NSS: 2 SGI LDPC
2021-05-18 08:58:25.951359+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_tx type=8 qos=0 multicast=0 len=24 subtype=0 qid=65535 using mgmt tid
2021-05-18 08:58:35.961183+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_tx type=8 qos=0 multicast=0 len=24 subtype=0 qid=65535 using mgmt tid
2021-05-18 08:58:45.968551+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: ieee80211_recv_deauth
2021-05-18 08:58:45.968553+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: Deauth received, reason 16
2021-05-18 08:58:45.968559+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_del_task iwx_ba_task
2021-05-18 08:58:45.968560+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_del_task iwx_htprot_task
2021-05-18 08:58:45.968586+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_add_task iwx_newstate_task
2021-05-18 08:58:45.968593+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: taskq task_add iwx_newstate_task add to queue
2021-05-18 08:58:45.968598+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: ieee80211_recv_deauth 3134 nstate=2
2021-05-18 08:58:45.968609+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_newstate_task
2021-05-18 08:58:45.968611+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_newstate_task sc->sc_flags & IWX_FLAG_SHUTDOWN false
2021-05-18 08:58:45.968620+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0xd1
2021-05-18 08:58:45.968621+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_disable_beacon_filter
2021-05-18 08:58:45.968627+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0xd2
2021-05-18 08:58:45.968956+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0xd1 done
2021-05-18 08:58:45.969827+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0xd2 done
2021-05-18 08:58:45.969846+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x28
2021-05-18 08:58:45.970136+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x28 done
2021-05-18 08:58:45.970140+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_phy_ctxt_cmd
2021-05-18 08:58:45.970143+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_phy_ctxt_cmd_uhb: 2ghz=0, channel=157, channel_width=2 pos=4 chains static=0x1, dynamic=0x1, rx_ant=0x3, tx_ant=0x3
2021-05-18 08:58:45.970147+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x8
2021-05-18 08:58:45.970756+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x8 done
2021-05-18 08:58:45.970761+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_disassoc
2021-05-18 08:58:45.970762+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd_pdu_status
2021-05-18 08:58:45.970767+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x18
2021-05-18 08:58:45.971048+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x18 done
2021-05-18 08:58:45.971054+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x1e
2021-05-18 08:58:45.971194+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x1e done
2021-05-18 08:58:45.971198+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd_pdu_status
2021-05-18 08:58:45.971203+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x18
2021-05-18 08:58:45.971343+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x18 done
2021-05-18 08:58:45.971350+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x19
2021-05-18 08:58:45.971600+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x19 done
2021-05-18 08:58:45.971605+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_deauth
2021-05-18 08:58:45.971606+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd_pdu_status
2021-05-18 08:58:45.971609+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x2b
2021-05-18 08:58:45.971996+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x2b done
2021-05-18 08:58:45.972010+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x28
2021-05-18 08:58:45.972183+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x28 done
2021-05-18 08:58:45.972187+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_auth
2021-05-18 08:58:45.972188+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_phy_ctxt_cmd
2021-05-18 08:58:45.972190+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_phy_ctxt_cmd_uhb: 2ghz=0, channel=157, channel_width=2 pos=4 chains static=0x1, dynamic=0x1, rx_ant=0x3, tx_ant=0x3
2021-05-18 08:58:45.972194+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x8
2021-05-18 08:58:45.973070+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x8 done
2021-05-18 08:58:45.973086+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd: sending command 0x28
2021-05-18 08:58:45.973257+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_cmd_done: command 0x28 done
2021-05-18 08:58:45.973262+0800 0x1ec      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_send_cmd_pdu_status
[... cut ...]
2021-05-18 08:59:17.622069+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: ieee80211_reset_scan
2021-05-18 08:59:18.568945+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_rs_update new rate: 0x4c00f017: VHT | ANT: AB BW: 80Mhz MCS: 7 NSS: 2 SGI LDPC
2021-05-18 08:59:57.718374+0800 0x163      Default     0x0                  0      0    kernel: (AirportItlwm) itlwm: iwx_rs_update new rate: 0x4c00d018: VHT | ANT: AB BW: 80Mhz MCS: 8 NSS: 2 NGI LDPC
```

williambj1, 2021-05-18:
CC @zxystd

richardkindom, 2021-05-19:
I met up with the same problem under 1.3.0.
It works fine after I rollbacked to 1.2.0 or upgraded to 1.4.0.

tmerten, 2021-06-14:
I just wanted to confirm that (for the last three hours or so) downgrading from `1.3.0` to `1.2.0` resolved this issue for me, too. Running an Asus PCE-AX3000 (the card is not detected running the open source version of the driver by the way).

tuancuong92, 2021-06-17:
Yeah I do confirm that 1.2.0 is better (my specs: AX201 & macOS 11.4)

williambj1, 2021-08-17:
Hi all, is this issue still present in the latest release?

alyxferrari, 2021-08-24:
@williambj1 I think it is. I'm using AirportItlwm v2.0.0 stable for Monterey and in macOS Monterey recovery, I can connect to my Wi-Fi network but after a couple seconds it either 1. disconnects and starts looking for another network or 2. all Wi-Fi networks disappear from the list (Wi-Fi is still on, but nothing appears). After they disappear they come back shortly and the cycle continues until eventually it connects continuously.

alyxferrari, 2021-08-25:
It's worth noting that after a couple minutes of the instability described earlier, the connection is pretty much perfect without any interruptions.

zxystd, 2021-08-25:
@alyxferrari No log so I don't know what is happening.

williambj1, 2022-03-03:
Closed due to no further response.

closed 2022-03-03 by williambj1