OpenIntelWireless/itlwm #599: Intermittent Slow boot - Intel firmware interrupt

madvinegar, 2021-05-19:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**
macOS Big Sur 11.3.1

- Kext Version: 1.3.0
- WiFi Card Model: intel dual band Wireless-AC 3168NGW (Stone Peak) 8086:24FB
- Product ID: 8086:24FB
- macOS Version: Big Sur 11.3.1

**Description**

Slow boot

**Bug Report Archive**
[  502.972368]: IntelFirmware: interrupt continue
[  502.972374]: IntelFirmware: HCI_OP_INTEL_VERSION
[  502.975408]: IntelFirmware: interrupt wait
[  502.976391]: IntelFirmware: Intel device is already patched. patch num: 34IntelFirmware: interrupt continue
[  502.976404]: IntelFirmware: HCI_OP_INTEL_EVENT_MASK
[  502.980388]: IntelFirmware: interrupt wait
[  502.981384]: IntelFirmware: interrupt continue
[  502.981390]: IntelFirmware: End download
[  502.981391]: IntelFirmware: Clean up...
[  502.981631]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981642]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981650]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981656]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981663]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981669]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981675]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981681]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981687]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981692]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981697]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981703]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981709]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981715]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981720]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981726]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981732]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981737]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981743]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981749]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981754]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981759]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981765]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981776]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981781]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981788]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981792]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981797]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981802]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981807]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981813]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981818]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981826]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981835]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981840]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981847]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981854]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981860]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981865]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981872]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981880]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981886]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981891]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981897]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981904]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981910]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)
[  502.981915]: IntelFirmware: onRead unhandle status (-536870165) operation was aborted)

**Kext Download Source**

Downloaded from official GitHub page

The slow boot does not happen every time. But when it happens it gets stuck at the intelfirmware section.
I also attach a photo of the verbose when booting. When I finally get into the desktop, the wifi is not working and I need to switch it off and on again to work.

[image]
[image]

zxystd, 2021-05-25:
I think it is an EFI issue? Seems that your device is not responding the HCI request, I don't know why.

BTW, You can remove the IntelFirmware.kext and use injector only to resolve the issue.

madvinegar, 2021-05-25:
Thank you for your reply! I was reading through the various issues, and I have seen that the injector kext was creating the problem. So I removed the Intelfirmware kext, but still I got the same slow boot.
Now I have removed both kexts and no more slow boots.

But I will also try what you said, i.e. remove the intelfirmware kext and install only the injector kext.

I will revert with any findings.
Thanks again.

zxystd, 2021-06-21:
@madvinegar Please try the latest v2.0.0 version of IntelBluetoothFirmware.kext. Anything update please go to that repo. I am going to close this issue, thanks.

closed 2021-06-21 by zxystd