OpenIntelWireless/itlwm #600: AX201 keeps disconnecting

taita87, 2021-05-19:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 2.0.0, earlier 1.2.0
- WiFi Card Model: AirPort Extreme  (0x8086, 0x74)
- Product ID: AX201
- macOS Version: Big Sur 11.1

**Description**
Wifi works but it randomly disconnects (OFTEN) and I have to manually reconnect selecting the network.
When using 1.2.0 I also had to reconnect manually, but often i cannot connect to the wifi network but got instead an exclamation mark on the wifi icon and had to turn it off and on several times before being able to reconnect. On 2.0.0 it seems I just got the disconnection issue without the exclamation mark problem.

**Bug Report Archive**
I'm using AirportItlwm but cannot open http://localhost:8080/itlwm/Troubleshooting.html#manually-for-airportitlwm

**Kext Download Source**
from official page here

zxystd, 2021-05-19:
https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#getting-kext-logs-for-loading-failures

please collect the log if error occurred.

taita87, 2021-05-19:
> https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#getting-kext-logs-for-loading-failures
> 
> please collect the log if error occurred.

uhm I don't think it failures to log the kext, if I open system information, under wifi i have 
"Firmware Version:	itlwm: 2.0.0 fw: 48.1335886879.0"
So I think the kext is loaded correctly

zxystd, 2021-05-20:
@taita87 Without log, I don't know what is happening. BTW, the log is the running log, It's not what you are thinking the loading log.
You can reopen the issue with the running log attached.

closed 2021-05-20 by zxystd