OpenIntelWireless/itlwm #601: Prelink injection AirportItlwm.kext () - Invalid Parameter

kleuter, 2021-05-19:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
No

**Environment**

- Kext Version: 1.3
- WiFi Card Model: 8265NGW
- Product ID: 
- macOS Version: Catalina

**Description**
opencore doesn't load AirportItlwm
kextstat reports that itlwm is loaded

[image]

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.3.0

vit9696, 2021-05-19:
You need SecureBootModel = Default for that to work. `prelinkedkernel` does not have IO80211Family.kext. You are best to provide the log with OpenCore DEBUG version, by the way.

vit9696, 2021-05-19:
Errng, and your config is really broken. Try running `ocvalidate` (bundled with OpenCore) on it, it will print many issues:

```
$ ./ocvalidate config.plist 
OCS: Missing key Base, context <Patch>!
OCS: Missing key BaseSkip, context <Patch>!
OCS: Missing key Base, context <Patch>!
OCS: Missing key BaseSkip, context <Patch>!
OCS: Missing key AllowRelocationBlock, context <Quirks>!
OCS: Missing key ForceBooterSignature, context <Quirks>!
OCS: Missing key ProvideMaxSlide, context <Quirks>!
OCS: Missing key Patch, context <Booter>!
OCS: Missing key Arch, context <Add>!
OCS: Missing key Arch, context <Add>!
OCS: Missing key Arch, context <Add>!
OCS: Missing key Arch, context <Add>!
OCS: Missing key Arch, context <Add>!
OCS: Missing key Arch, context <Add>!
OCS: Missing key Arch, context <Add>!
...
```

kleuter, 2021-05-21:
> the immutablekernel should be used
how to enable it?

vit9696, 2021-05-21:
I mean EfiBoot should use it in this case. Not sure why it does not, maybe something with NVRAM.

kleuter, 2021-05-21:
Fixed all config errors/warnings, logs shows success loading of kext, but no wifi.
[opencore-2021-05-21-204504.txt](https://github.com/OpenIntelWireless/itlwm/files/6524991/opencore-2021-05-21-204504.txt)
[config.plist.txt](https://github.com/OpenIntelWireless/itlwm/files/6524993/config.plist.txt)

vit9696, 2021-05-21:
itlwm.kext is not AirportItlwm.kext, it does not support native Wi-Fi interface but requires HeliPort app. You should rather replace it with AirportItlwm.kext.

closed 2021-05-22 by kleuter