OpenIntelWireless/itlwm #602: AX200 不时断连且速度极低只有10~100Kbps

Devifish, 2021-05-20:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version:  v2.0.0-alpha (2021-05-17)
- WiFi Card Model: AX200 (Router Asus AX86U)
- Product ID: Unknow
- macOS Version: 11.3 (OpenCore 0.6.9)

**Description**

能够连接WIFI使用，但是使用一段时间后会自动断连而且速率极低只有10~100Kbps 在160mhz连接速率显示1200M（也在40m 80m测试过，问题一样）内网延迟也在100ms以上。 打开百度也需要加载10S以上, 详情如下图。
[image]
[image]

**Bug Report Archive**

[Report_2021-05-20_12-59-35.log](https://github.com/OpenIntelWireless/itlwm/files/6513046/Report_2021-05-20_12-59-35.log)
[Log_2021-05-20_13-00-00.log](https://github.com/OpenIntelWireless/itlwm/files/6513047/Log_2021-05-20_13-00-00.log)

**Kext Download Source**

[AirportItlwm-Big_Sur-v2.0.0-DEBUG-alpha-33f4fbb.zip](https://github.com/OpenIntelWireless/itlwm/files/6513048/AirportItlwm-Big_Sur-v2.0.0-DEBUG-alpha-33f4fbb.zip)

Devifish, 2021-05-20:
忘了说，在使用itlwm时没有以上问题(仅AirportItlwm)  速率可以达到200MB/s

zxystd, 2021-05-25:
Know issue, WiFi card will definitely decrease throughput if it is doing scanning work, and out current AirportItlwm framework will do long time background scanning if the airportd need it, and behave as the unstable network speed.

zhangdp, 2021-06-16:
@zxystd 您好，我并不赞同"not have too much benefits on experience"，正是因为AirportItlwm比较"原生"且不需要额外安装软件，我们才喜欢用它超过于itlwm.kext，但是却因为这个问题导致AirportItlwm基本不可用，体验很差

win1010525, 2021-06-16:
> @zxystd 您好，我并不赞同"not have too much benefits on experience"，正是因为AirportItlwm比较"原生"且不需要额外安装软件，我们才喜欢用它超过于itlwm.kext，但是却因为这个问题导致AirportItlwm基本不可用，体验很差

你不赞同解决方案不代表没有替代品，itlwm在调试和速度上都有明显的优越性，而airportitlwm则需要一直重启。
网卡扫描掉速原本就是无解，这是硬件缺陷，只是macOS的扫描过于频繁导致问题更严重了，你可以买一台MacBook Pro和iPhone 12，他们都有扫描掉速，只是扫描很快罢了。
当然，如果你对此有何疑问的话，你可以自己clone仓库，修改代码，然后pr修复这个问题。

zzp1024, 2021-07-08:
希望能解决~~只是希望~~~

samleong123, 2021-08-03:
貌似升级 Big Sur 11.5.1  后我的airportitlwm + AX200 速度回来了
[image]

williambj1, 2021-10-02:
@Devifish 日志中没看到跟路由器断连。Airport limitations are mentioned above.