OpenIntelWireless/itlwm #603: AirportItlwm Not Loading After Updating Kext

ehinium, 2021-05-20:
**Have You Read Our Docs**
- Yes

**Are You Reporting A Bug**
- Yes

**Environment**

- Kext Version: 1.1.0, 1.2.0, 1.3.0, 2.0.0 (Happened everytime I have updated)
- WiFi Card Model: AX201 Wi-Fi 6 (802.11 ax (2x2))
- Product ID: 
- macOS Version: BigSur 11.3.1

**Description**
- This happens every time I update the kext; when I add the new version to my EFI folder and reboot, the AirportItlwm and also IO80211Family does not load at all.
This was randomly fixed previous times but not this time(v1.3.0). Apple Secure boot is on Default, and I've already tried resetting NVRAM and network.
I tried forcing IO80211Family to load as documents said but It didn't help:
[image]

[image]

**Bug Report Archive**
<a href="https://pastebin.com/HkGwEpFt">Logs For Loading Failure</a>

**Kext Download Source**
<a href="https://github.com/OpenIntelWireless/itlwm/releases/download/v1.3.0/AirportItlwm_v1.3.0_stable_BigSur.kext.zip">AirportItlwm_v1.3.0_stable_BigSur.kext</a>

williambj1, 2021-05-20:
Does itlwm (not AirportItlwm) work as intended?

ehinium, 2021-05-20:
> Does itlwm (not AirportItlwm) work as intended?

I hadn't tested itlwm before because Airportitlwm was fine. I tried it now, and I faced a kernel panic, so I switched back to Airportitlwm to boot my device and fix the KP, and now it's loaded and working! I don't have any idea why this is happening every time and fixing out of nowhere.

williambj1, 2021-05-20:
Please attach the panic log

ehinium, 2021-05-20:
I don't have the log at the moment unfortunately. The panic error was "image kexts\itlwm.kext\contents\macos\itlwm is missing for injected kext itlwm.kext" at the very beginning of boot process.
I can add it again and send you the log if you need?

ehinium, 2021-05-21:
> Please attach the panic log

I figured out how to fix this finally.
Before upgrade or downgrade both AirportItlwm or Itlwm, I deleted the kext from OC/Kexts and config.plist and reboot.
Then I added the kext to my OC/Kexts and config.plist, and after rebooting again, it works without any issue. Forcing IO80211Family is not needed for me.

I have another issue with v1.3.0 and v2.0.0. It can't connect to one of my modems (self-assigned IP error with a "!" on wifi logo in menubar). Should I open a new issue for it?

zxystd, 2021-05-25:
User mistakes. If you found another problems, please open another issue if nobody had report the same problem here. Thanks.

closed 2021-05-25 by zxystd