OpenIntelWireless/itlwm #605: Advanced-N 6205 doesn't search networks after off state.

KlausWoof, 2021-05-26:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm and itlwm 1.3.0
- WiFi Card Model: Intel Advanced-N 6205
- Product ID: 85
- macOS Version: 10.14.6

**Description**
When I press Turn Wi-Fi Off or connect to other WiFi network, I can't switch back, because my wifi adapter doesn't find any networks.

**Bug Report Archive**

[Report_2021-05-26_12-29-36.log.zip](https://github.com/OpenIntelWireless/itlwm/files/6545153/Report_2021-05-26_12-29-36.log.zip)

**Kext Download Source**

Official 1.3.0 build

zxystd, 2021-05-26:
The log is wrong, can you collect again?
- Add `msgbuf=1048576` into `boot-args` using either nvram or your bootloader's config
- Run the following command in terminal:
   - `sudo dmesg | grep -i "itlwm" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log`

A log file will then be created on your Desktop.

KlausWoof, 2021-05-26:
The new log file attached.

KlausWoof, 2021-05-26:
The new log file attached.
[Log_2021-05-26_14-15-22.log.zip](https://github.com/OpenIntelWireless/itlwm/files/6545905/Log_2021-05-26_14-15-22.log.zip)

pigworlds, 2021-05-26:
I am unable to reproduce this issue on my Intel Advanced-N 6205 setup. I think I have spotted the issue in the log and created a fix PR.
@KlausWoof, will you able to try this build
[Airportitlwm-Mojave.zip](https://github.com/OpenIntelWireless/itlwm/files/6548960/Airportitlwm-Mojave.zip) to see if it works for you now?

KlausWoof, 2021-05-26:
[Archive.zip](https://github.com/OpenIntelWireless/itlwm/files/6549347/Archive.zip)

I'm testing the attached kext and for short distance it came back to work from off state :) 
Some of logs attached.

zxystd, 2021-06-03:
@KlausWoof Hi, is it solved?

KlausWoof, 2021-06-03:
@zxystd Hi! Yes!

zxystd, 2021-06-03:
> @zxystd Hi! Yes!

Ok, thanks. If there is something wrong happened, please open another issue. And this issue is going to be closed.

closed 2021-06-03 by zxystd