OpenIntelWireless/itlwm #607: AC 9462 连不上AP

nu11plex, 2021-05-27:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: v1.3.0 Stable
- WiFi Card Model: Intel WIreless-AC 9462 (WiFi Model TP-Link WR842N v4.0)
- Product ID: Unknown
- macOS Version: 10.15.7 (OpenCore 0.6.3)

**Description**
不能连接wifi，如下图
[image]
Log显示"itlwm: iwm_rx_tx_cmd_single 1337 OUTPUT_ERROR status=142"

**Bug Report Archive**
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/6554802/itlwm.log)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v1.3.0/AirportItlwm_v1.3.0_stable_Catalina.kext.zip

williambj1, 2021-05-27:
Logs incomplete, please read the docs and adjust the log buffer size.

Try itlwm instead of AirportItlwm as well.

nu11plex, 2021-05-28:
So how to get the full log of the AirportItlwm?
I'm using sudo dmesg | grep "AirportItlwm"

kpostekk, 2021-05-29:
I have similar problem. In my case macOS has passphrase for network, because it is in my iCloud.

Environment

- Kext Version: v2.0.0-alpha
- WiFi Card Model: Intel WIreless-AC 9462 (Builtin to MB _AORUS Z490 ELITE AC_)
- Product ID: Unknown
- macOS Version: 11.4 (OpenCore 0.6.9)

[AirItlDebug_2021-05-29_11-50-20.zip](https://github.com/OpenIntelWireless/itlwm/files/6564421/AirItlDebug_2021-05-29_11-50-20.zip) _w/ adjusted buffer size_

nu11plex, 2021-05-29:
Here the new logs.   
[AirItlDebug_2021-05-30_00-59-29.zip](https://github.com/OpenIntelWireless/itlwm/files/6565036/AirItlDebug_2021-05-30_00-59-29.zip)

zxystd, 2021-05-30:
@nu11plex 
The log is wrong, can you collect again?
- Add `msgbuf=1048576` into `boot-args` using either nvram or your bootloader's config
- Run the following command in terminal:
   - `sudo dmesg | grep -i "itlwm" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log`

A log file will then be created on your Desktop.

zxystd, 2021-05-30:
@kpostekk Have you tried another router?

kpostekk, 2021-05-30:
With Bluetooth disabled I have tried to connect to iPhone hotspot and my older access point:

- I was able to connect with iPhone but I couldn't acquire response with IP address
- Connecting to my older AP ended similarly like connecting to my main WiFi

_PS. my main WiFi is a mesh (2x TP LINK Deco M5)_

zxystd, 2021-05-30:
@kpostekk Try the latest alpha please.

kpostekk, 2021-05-30:
Looks like it partially works for me.

**Connecting with iPhone: success**

[image]

**Connecting with mesh network: failure**

I have tried reentering password, but it didn't helped. 

Logs are here: 
[AirItlDebug_2021-05-30_22-55-37.zip](https://github.com/OpenIntelWireless/itlwm/files/6566899/AirItlDebug_2021-05-30_22-55-37.zip)

**Connecting to regular router: success**
[image]

JBramauer, 2021-05-31:
Similar issue here when I tried itlwm 1.3.0 stable a couple of days ago. I'll collect a log when I have time to. For now I can add the additional Info that my 2.4 GHz and 5 GHz Band networks have the same SSID.

Board: Gigabyte Z490 Aorus Elite AC with 9462 WiFi.

kpostekk, 2021-06-01:
@JBramauer consider updating your EFI and latest alpha of Airportitlwm should fix some of your issues.

zxystd, 2021-06-01:
@kpostekk Can you tried this one? 
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6574366/itlwm.kext.zip)

kpostekk, 2021-06-01:
I could easily connect to 5 GHz network, but still not connecting with my mesh. (Ofc I had disabled AirportItlwm and installed HeliPort)
Logs: 
[bugreport_17031.zip](https://github.com/OpenIntelWireless/itlwm/files/6579900/bugreport_17031.zip)

zxystd, 2021-06-02:
> I could easily connect to 5 GHz network, but still not connecting with my mesh. (Ofc I had disabled AirportItlwm and installed HeliPort)
> Logs:
> [bugreport_17031.zip](https://github.com/OpenIntelWireless/itlwm/files/6579900/bugreport_17031.zip)

you mean you can connect 11ac network before? weird issue, maybe your mesh router doesn't support LDPC?

kpostekk, 2021-06-02:
> you mean you can connect 11ac network before? 

I mean, after installing your kext I was able to do it. Previously, connecting to some 5 GHz has some issues (while using AirportItlwm)

> weird issue, maybe your mesh router doesn't support LDPC?

Is there any easy way to check it? I'm using Tp-Link Deco M5. It was released in 2019 so it's pretty new.

zxystd, 2021-06-02:
maybe you can check it in router setting page? hmmm, I will compile another one which block LDPC support kext for you.
thank you for the test!

kpostekk, 2021-06-02:
> maybe you can check it in router setting page?

I wish I had one lmao. The best I can do is to browse basic settings in a mobile app. So, sorry, I can't really check that.

JBramauer, 2021-06-02:
Alright, I just updated to OC 0.6.9 - issue persists.
itlwm 1.2.0 works fine
latest kext you posted in this thread does not connect.

I'm attaching a log of following sequence:
boot -> connection fail to known network -> remove network in heli port -> reconnect (re-enter password) -> fail

[itlwmlog.txt](https://github.com/OpenIntelWireless/itlwm/files/6587954/itlwmlog.txt)

Sorry, I know this is not your standard bug report format and I will re-do this in the next couple of days.
It got really late tonight updating OC ;)

zxystd, 2021-06-03:
@kpostekk @JBramauer Please try the latest alpha version I just push to GitHub.

kpostekk, 2021-06-03:
[image]

There weren't assets so I have compiled them.
[Big Sur.zip](https://github.com/OpenIntelWireless/itlwm/files/6592759/Big.Sur.zip)
[Catalina.zip](https://github.com/OpenIntelWireless/itlwm/files/6592761/Catalina.zip)

JBramauer, 2021-06-03:
Still no luck, new log:

[airportitlwm2alphalog.txt](https://github.com/OpenIntelWireless/itlwm/files/6593970/airportitlwm2alphalog.txt)

zxystd, 2021-06-04:
> Still no luck, new log:
> 

@JBramauer @kpostekk  oh, sorry, that's my mistake, I miss something when commit the changes. Now it is updated, please try again.

kpostekk, 2021-06-04:
I have compiled latest commit and once I was able to even connect with my mesh. The bad news is that i'm not sure I have logged that. Anyway there are logs:  
[AirItlDebug_2021-06-04_12-06-40.zip](https://github.com/OpenIntelWireless/itlwm/files/6597724/AirItlDebug_2021-06-04_12-06-40.zip)

JBramauer, 2021-06-06:
Alright, here's the update:

I was able to connect to my WiFi once, then everything was fine for a couple of minutes, until the connection dropped and I could not reconnect.

Log from boot until this point in time:
[alphalog.txt](https://github.com/OpenIntelWireless/itlwm/files/6604655/alphalog.txt)

The connection issues persisted after a reboot, so I was not able to connect to my wifi ever again after the first connection drop.

JBramauer, 2021-06-09:
Are there any relevant changes regarding this issue in the latest alpha build?

nu11plex, 2021-06-12:
Thx everybody,
I just tried the new alpha version of AirportItlwm.
It works fine.
Finally I can use the native macOS wifi.

closed 2021-06-12 by nu11plex

PKRN0, 2021-06-12:
Hi, I had the similar problem and was watching the process throughout here.
However,I still have the problem with the latest commit.

Environment:

- WLAN Card: AC 9462 (Built in to Samsung NT350XCR laptop)
- Kext Version: AirportItlwm-BigSur 2.0.0 alpha
- OS Version: macOS Big Sur 11.4 with OC0.6.9
- Product ID: Unknown
- Router Model: Iptime A604M

When I connect to the wifi, it connects. After 1~2sec, it disconnects. It just loops reconnecting and disconnecting.
It works well with AirportItlwm 1.2.0. On ver1.3.0, it did not work.

Log
[Log_Aitlwm-alp-0612.log](https://github.com/OpenIntelWireless/itlwm/files/6641814/Log_Aitlwm-alp-0612.log)

JBramauer, 2021-06-14:
This should be re-opened. Issue still exists in latest alpha.