OpenIntelWireless/itlwm #612: Native Wifi

tahaklkt-osxinfo, 2021-06-02:
Hello. I have Intel Centrino Advanced-N 6235. I want activate this but Heliport is have issues. So I want native wifi. How can I make this ? Im using opencore 0.6.9 and macOS Catalina. I tried nstall AirportItlwm.kext to L/E but this is not worked for me it seems a Ethernet. How can I fix this ? Thanks !

zxystd, 2021-06-02:
airportitlwm is alteady a native wifi driver, can you capture the screentshot?
BTW, please fill the full info according to the issue template.

tahaklkt-osxinfo, 2021-06-02:
Sorry. I have not debug files. Im copying airportitlwm to oc kext or L/E but its seems as Ethernet. I can only activate itlwm + heliport

zxystd, 2021-06-03:
@tahaklkt-osxinfo 
- Add `msgbuf=1048576` into `boot-args` using either nvram or your bootloader's config
- Run the following command in terminal:
   - `sudo dmesg | grep -i "itlwm" > ~/Desktop/Log_"$(date '+%Y-%m-%d_%H-%M-%S')".log`

A log file will then be created on your Desktop.

zxystd, 2021-06-03:
@tahaklkt-osxinfo If you are using AirportItlwm.kext and it worked, it is impossible to attach as an Ethernet interface, I  think you was confused by the interface's name "Ethernet"

closed 2021-06-04 by zxystd