OpenIntelWireless/itlwm #614: [AirportItlwm] Panic after long sleep on AX200

3xpl0itz, 2021-06-06:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
Kext Version: AirPortItlwm v2.0 (Alpha)
WiFi Card Model: Intel AX200
Product ID: 0x8086, 0x84
macOS Version: macOS Big Sur 11.4

**Description**
After my hack sleeps for a long period of time, I notice that it reboots, and I am greeted with the log linked below.

**Bug Report Archive/Logs**
https://pastebin.com/s18rkuFX

**Upload your Kext**
Using the latest AirportItlwm v2.0 alpha commit.

zxystd, 2021-06-07:
Is it reproducible?

3xpl0itz, 2021-06-07:
I believe it has happened more than once under periods of prolonged sleep.

blacksd, 2021-09-11:
Came here to report the same exact symptoms, but with an AC 3165. Short sleep periods (up to 30 minutes) are fine, leave the laptop overnight, Kernel Panic.

williambj1, 2021-10-02:
The device is completely dead after entering sleep mode, it did not respond in 3 minutes and Apple's kext triggered the panic. There is a quirk in OpenCore that disables this check, but your hack will likely stop functioning as well.

Seems to be related to hardware, EFI, and ACPI, nothing could be done in the driver's side.

closed 2021-10-02 by williambj1