OpenIntelWireless/itlwm #616: Boot loop when booting macos in verbose mode

naveenkrdy, 2021-06-07:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: latest [2.0.0 alpha](https://github.com/OpenIntelWireless/itlwm/commit/d97c0d23f5f52840282da4858de6e0800c1fdb7c)
- WiFi Card Model: AX200
- Product ID: 0x8086, 0x1654
- macOS Version: 11.4

**Description**
just updated to latest alpha and if i boot macos in verbose mode (-v) then it just boot loops with same message again and again as shown in below pic.

 
[image]

t0niX, 2021-06-08:
Same with OpenCore 0.7.0 and AirportItlwm_v1.3.0_stable_BigSur.kext.zip.

Happened to me the first time after updating to OpenCore 0.7.0 today. After lots (~50) of loops MacOS finished booting though.

naveenkrdy, 2021-06-15:
seems to be fixed now with latest commits 
thanks !

closed 2021-06-15 by naveenkrdy

t0niX, 2021-06-15:
This is not fixed with the latest official build. You might want to release a hotfix if you don’t plan to rollout a major update soon.

t0niX, 2021-06-19:
I can confirm that the issue is not fixed in the latest alpha as well. I still get the boot loop.

Can you please re-open the issue - thanks.

zxystd, 2021-06-19:
> I can confirm that the issue is not fixed in the latest alpha as well. I still get the boot loop.
> 
> Can you please re-open the issue - thanks.

what's your card?

t0niX, 2021-06-20:
I‘m using an ASUS Z390 XI Maximus Hero Wifi.

The card is an Intel 9560.

zxystd, 2021-06-20:
@t0niX I am rewriting the whole IntelBluetoothFirmware driver, although it is still in process, can you test it?
[IntelBluetoothFirmware.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6681630/IntelBluetoothFirmware.kext.zip)