OpenIntelWireless/itlwm #617: macOS 12 Monterey Support?

AvaQueen-Official, 2021-06-07:
Just wondering if we'll need a separate Airportitlwm.kext for WiFI to work on macOS 12 :)

zxystd, 2021-06-08:
[AirportItlwm-Monterey.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6613211/AirportItlwm-Monterey.kext.zip)
Try this one and attach the running logs. Thanks.

vannadium1788, 2021-06-08:
> [AirportItlwm-Monterey.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6613211/AirportItlwm-Monterey.kext.zip)
> Try this one and attach the running logs. Thanks.
On version 12 (21A5248p) with your kext, the system sees available wi-fi networks, but cannot connect to any of them.

zxystd, 2021-06-08:
@vannadium1788 
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6614897/AirportItlwm.kext.zip)
what about this one, can it recognize the wifi encryption algorithm correctly?

I have not install macOS 12, so I need others help for testing.

vladdenisov, 2021-06-08:
I've tried this version, its also didn't work for me
[log_stream.txt](https://github.com/OpenIntelWireless/itlwm/files/6616055/log_stream.txt) - this is log stream when I was trying to connect to my network

AvaQueen-Official, 2021-06-08:
@zxystd when I use the beta Airportitlwm-Monterey.kext I get this:
[image]

Also having issues with Bluetooth:
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/295

vannadium1788, 2021-06-08:
> @vannadium1788
> [AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6614897/AirportItlwm.kext.zip)
> what about this one, can it recognize the wifi encryption algorithm correctly?
> 
> I have not install macOS 12, so I need others help for testing.

[image]
I tried it but unfortunately the problem persists

seungjunProgramming, 2021-06-08:
I’m having the exact same issue with @vladdenisov 
it says WEP, but my network isn’t WEP.
Also, It’s not connecting to the internet ethier.

AvaQueen-Official, 2021-06-08:
Yeah this seems to be a mutual issue.

iTer99, 2021-06-09:
I have to problem, it show available network but can't connect to them

kidonng, 2021-06-09:
I can confirm AirportItlwm works on both 2.4Ghz and 5GHz **open** network, whether they are from router or hotspots. As for WPA protected networks they are recognized as WEP (refer to conversations above) and thus not connecting.

Some other minor issues:
- Scanning seems to not work reliably
- (Somehow) degraded performance

bfitzgit23, 2021-06-09:
> I can confirm AirportItlwm works on both 2.4Ghz and 5GHz **open** networks, whether they are from router or hotspots. As for **WPA protected** networks they are recognized as WEP (refer to conversations above) and thus not connecting.
> 
> Some other minor issues:
> 
> * Scanning seems to not work reliably
> * (Somehow) degraded performance

yeah im getting the same issue on my lemur system76 laptop.

BlackOtton, 2021-06-09:
I think the best thing to get WiFI to work for now on Monterey is through itlwm.kext and Heliport.
Connect's to WPA2 protected 5Ghz network without any problem.
[image]
[image]

It might not be perfected, however it's still better than not having network at all.

b33b, 2021-06-09:
> I think the best thing to get WiFI to work for now on Monterey is through itlwm.kext and Heliport.
> Connect's to WPA2 protected 5Ghz network without any problem.
> [image]
> [image]
> 
> It might not be perfect, however it's still better than not having network at all.

I confirm! Indeed, if you take the latest version of itwm and Heliport, then wifi will work properly.
Thanks to Heliport, wpa2 (2.4GHz) authentication is successful, and as a temporary option, it is a bang. Please note, the built-in utility does not see the wifi network
[image]

AvaQueen-Official, 2021-06-09:
how fast/reliable is itlwm compared to airportitlwm? I don't like the fact that it's non-native... @b33b @BlackOtton

b33b, 2021-06-10:
> how fast/reliable is itlwm compared to airportitlwm? I don't like the fact that it's non-native... @b33b @BlackOtton

In terms of the speed of work, it still pleases. I can't say anything about stability, since testing takes time. But in two sessions, the network did not fall off, but worked like a native utility.
However, do not forget that this is a temporary version and you need to wait for the normal version of airportitwm from @zxystd

liyulong214, 2021-06-11:
感谢你们的工作，期待解决！是否开放赞助？愿意聊表心意

williambj1, 2021-06-11:
Supported in 7214459c1a35abb343171ae990227ccf352b69cd

closed 2021-06-11 by williambj1

samwzlim, 2021-06-11:
Does handoff, universal clipboard etc work on Monterey as it did in Big Sur using the latest alpha release?

williambj1, 2021-06-11:
> Does handoff, universal clipboard etc work on Monterey as it did in Big Sur using the latest alpha release?

No, because Bluetooth is broken. Non-native Bcrm cards have the same issue as well.

webdeveloper1877, 2021-11-06:
Hi,if you are interested I have created the distribution (opencore 0.7.5)with all updated Kexts for "(macOS Monterey 12.0.1 (21A559)" practically everything that works in BigSur also works on Monterey, I had tried to install monterey with the distribution downloaded from your repository the (0.7.3) but it was not successful, after the first reboot it asked me to update the firmware of the ssd and the installation was blocked, while with OpenCore 0.7.5 macOS Monterey installs without problems and it works both bluetooth and wifi!

the ThinkPad model I tested it on is…

  | Thinkpad T460s 20F90002**
Processor | Core i5-6300U (2C, 2.4 / 3.0GHz, 3MB)
Graphics | Integrated Intel HD Graphics 520
Memory | 4GB Soldered + 16GB DIMM 2133MHz DDR4, dual-channel
Display | 14" Full HD (1920x1080) IPS, TouchScreen
Storage | Samsung 512GB SATA SSD.
Ethernet | Intel Ethernet Connection I219-LM (Jacksonville)
WLAN + Bluetooth | 11ac+BT, Intel® Dual Band Wireless-AC 8265, 2x2 card
Camera | HD720p resolution, low light sensitive, fixed focus
Audio support | HD Audio, Realtek ALC3245 codec, stereo speakers 1Wx2, dual array microphone, combo audio/microphone jack
Keyboard | 6-row, spill-resistant, multimedia Fn keys, LED backlight
Battery | Front Li-Polymer 3-cell (23Wh) and rear Li-Ion 3-cell (26Wh), both Integrated

link from where to download the distribution EFI OC 0.7.5 for macOS Monterey and Big Sur:

https://mega.nz/file/gsxARaQD#6qiPpIpkGvNcJfBUYXl7Hn19_PLEmUW4An06HQjAQDM