OpenIntelWireless/itlwm #618: Find it hard to connect a 802.1x network via a profile

richardkindom, 2021-06-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
MacOS 10.15.7

- Kext Version: 1.3.0Stable
- WiFi Card Model: 8260
- Product ID: 0x1130
- macOS Version: 10.15.7

**Description**
Find it hard to connect a 802.1x network via a profile. Sometimes connect/disconnect frequently to a 802.1x network after sleep.

**Bug Report Archive**
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/6612655/itlwm.log)

**Kext Download Source**
On official GitHub page.

richardkindom, 2021-06-08:
BTW: network name is Office and its MAC address has been shadowed due to privacy consideration.

richardkindom, 2021-06-09:
Hello, after several trials, I found out this issue was related to v1.3.0 and above. So I guess this problem relied on implementation of frequency selection? Or something like this?
After all I rolled back to 1.2.0. And it works perfectly

williambj1, 2021-06-14:
CC @zxystd

zxystd, 2021-07-30:
@richardkindom I had tried v2.0.0-alpha with gen1 cards, and it works fine, have you try that?

richardkindom, 2021-09-17:
> @richardkindom I had tried v2.0.0-alpha with gen1 cards, and it works fine, have you try that?

hey, finally I found out this issue was related to authentication hardware, not airportitlwm. So I would close this issue. Thanks for ur effort.

closed 2021-09-17 by richardkindom