OpenIntelWireless/itlwm #621: AX200 Unstable Speed with <10Mbps

samleong123, 2021-06-09:
**Have You Read Our Docs**

Yes

**Are You Reporting A Bug**

Yes

**Environment**

- Kext Version: 2.0.0 Alpha 
- WiFi Card Model: AX200
- Product ID: I don't know where to find it
- macOS Version: 11.4

**Description**

Unstable speed with speed below 10Mbps all the time 
My ISP provide me a 500Mbps synchronous speed
I can gain full speed via my iPad connected to the same 5GHz network and also ethernet to my computer
iPad : 
[image]
Ethernet : 
[image]
But when i connect to my computer , i can only get <10Mbps speed 
[image]
[image]

**Bug Report Archive**

[AirItlDebug_2021-06-09_11-30-15.zip](https://github.com/OpenIntelWireless/itlwm/files/6620920/AirItlDebug_2021-06-09_11-30-15.zip)

**Kext Download Source**

Github Releases , 9af1571

williambj1, 2021-06-09:
Did you try itlwm instead of AirportItlwm?

samleong123, 2021-06-09:
itlwm work perfectly 
[image]

so currently AirportItlwm will be more unstable than itlwm?

williambj1, 2021-06-09:
AirportItlwm does a scan whenever airport requests it to do so. As a result, AirportItlwm scans much more frequently than itlwm, which significantly slows down the network speed, but there might be other issues involved as well.

zxystd, 2021-06-15:
duplicate with #602

closed 2021-06-15 by zxystd