OpenIntelWireless/itlwm #625: itlwm detects network, but can't connect, with some routers.

andrsn0w, 2021-06-12:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: Tried both itlwm 2.0.0 and 1.3.0
- WiFi Card Model: Intel® Centrino® Advanced-N 6205
- Product ID: idk
- macOS Version: macOS Catalina 0.15.7 

**Description**
Unable to connect to certain networks. The D-LINK DAP-1155 works perfectly, but it is not possible to connect to the Huawei B593s-22 network. Both networks are 802.11b / g / n. HeliPort finds all networks, but when I try to connect to a Huawei WiFi network, it only lets me enter a password and then doesn't do anything.  I know that Advanced-N 6205 is experimental, but it works with DAP-1155 so i think it's a bug. I wi
ll include WiFi configuration settings screenshots too.
[image]
[image]

**Bug Report Archive**
[bugreport_23157.zip](https://github.com/OpenIntelWireless/itlwm/files/6642354/bugreport_23157.zip)
**Kext Download Source**
Kext: https://github.com/OpenIntelWireless/itlwm/releases
HeliPort(just in case):https://github.com/OpenIntelWireless/HeliPort/releases/tag/v1.4.0

andrsn0w, 2021-06-12:
itlwm: : missed beacon threshold set to 30 beacons, beacon interval is 100 TU
itlwm: ieee80211_recv_assoc_resp 3098 nstate=4
itlwm: : fatal firmware error
itlwm: firmware error log:
itlwm:   error type      = "UNKNOWN" (0x00002078)
itlwm:   program counter = 0x00009758
itlwm:   source line     = 0x0000063D
itlwm:   error data      = 0x4000801500006500
itlwm:   branch link     = 0x0000962A0000962A
itlwm:   interrupt link  = 0x0000D6BE00000000
itlwm:   time            = 1038049847

andrsn0w, 2021-06-12:
I was pretty bored and tried turning b/g/n modes, i found the issue:  802.11 b, g, and b/g works but b/g/n does not

andrsn0w, 2021-06-12:
Intel(R) Centrino(R) Advanced-N 6205 ABG
Intel(R) Centrino(R) Advanced-N 6205 BG
i read Compatibility page again and found that N does not work, but is there some way to disable N from itlwm side? I just want to be able to connect to every WiFi without WiFi reconfiguration

pigworlds, 2021-06-13:
The Centrino Advanced-N 6205 support was added in 1.2.0 stable (N with 20MHz channel). Will you able to give a try on the 1.2.0 stable?

In addition, will you able to try other channels like channel 6, or any 5GHz channels?

pigworlds, 2021-06-13:
I got able to reproduce this issue if the router is enabled 'greenfield' mode. Since the 'greenfield' is not properly implemented in 'iwn', it is better to disable it now.

Will you able to give a try this [itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/6643249/itlwm.kext.zip) ?

andrsn0w, 2021-06-13:
Hi! 
This Kext worked well with my HP ProBook 6470b, everything is fine. 
Probably this "Greenfield mode" was the issue.
I hope this "greenfield" mode will be properly implemented in the future.
Issue is still there, so i won't close this yet.

pigworlds, 2021-06-13:
Glad the purposed fix is working properly for you. I would consider the issue is fixed.

andrsn0w, 2021-06-13:
For me, issue is fixed, but issue is still there in itlwm versions, it should be fixed. But you already made pull request, so i will close this. Thanks again!

closed 2021-06-13 by andrsn0w