OpenIntelWireless/itlwm #628: AC9461 wifi on-off looping and disconnecting

Pl1y, 2021-06-15:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.2.0 and 1.3.0
- WiFi Card Model: Intel Wireless-AC 9461 Wireless Network Adapter
- PCI Product ID: PCI\VEN_8086&DEV_34F0&SUBSYS_02648086 and PCI\VEN_8086&DEV_34F0&CC_0280 
[image]

- macOS Version: 11.4 Big Sur

**Description**
When I connect to any of my two Wi-Fi networks, there are 3 possible outcomes:

1. the Wi-Fi connects, it says that everything is good, and then, after approx. 10 seconds or when I open something that needs internet, it loses the connection and stops showing the networks, sometimes it also says that Wi-Fi is off, then it enables Wi-Fi again and starts again from the beginning.
2. the Wi-Fi connects and stays connected, but it says that the internet isn't available
3. it tries to connect and pops up with the password window even though the network is saved and says: "Connection failed."

**Bug Report Archive**
([bugreport_51961.zip](https://github.com/OpenIntelWireless/itlwm/files/6656246/bugreport_51961.zip)

**Kext Download Source**
GitHub, releases page, itlwm_v1.3.0_stable.kext.zip and itlwm_v1.2.0_stable.kext.zip
Note: It happens with the same versions of AirportItlwm, too (I only used one at a time but I tried all four)

zxystd, 2021-06-15:
Have you tried v2.0.0-alpha?

Pl1y, 2021-06-15:
I have just now and its stuck on "calling mpo_policy_init for". [Here's the image of the panic.](http://imgur.com/a/L2slmgy)

zxystd, 2021-06-16:
> I have just now and its stuck on "calling mpo_policy_init for". [Here's the image of the panic.](http://imgur.com/a/L2slmgy)

itlwm will never trigger such kind of error, it is most likely a user mistake.

Pl1y, 2021-06-16:
The kext probably corrupted in the copying process because I transferred it from my phone, now I tried using a Linux boot USB and downloading from there and it still does it BUT after 3 minutes of trial and error, it finally connects once and it works after until I restart.

JBramauer, 2021-06-17:
Sounds a lot like what's happening on the 9462 chips, probably related?
https://github.com/OpenIntelWireless/itlwm/issues/607
(Issue closed, even though it shouldn't be...)

zxystd, 2021-06-18:
@Pl1y If the v2.0.0-alpha version is still having problems, please attach log, we don't know what is happening without log.

kpostekk, 2021-06-20:
[AirItlDebug_2021-06-20_12-23-00.zip](https://github.com/OpenIntelWireless/itlwm/files/6682107/AirItlDebug_2021-06-20_12-23-00.zip)

As far I remember from #607, only some network types were affected. For me 2.4 and 5GHz standalone routers work fine, but not mesh network.

zxystd, 2021-07-23:
Seems that there is really exists problems on AC9461/9462, but I don't know what's wrong, so I need more online testing and logs feedback(I don't have neither of the two cards), is anyone can help here?

kpostekk, 2021-07-23:
I can send some logs.

zxystd, 2021-07-23:
I think I got it working now, please check and test. @kpostekk @JBramauer @Pl1y 
[fix.zip](https://github.com/OpenIntelWireless/itlwm/files/6868076/fix.zip)

ultinux, 2021-07-23:
@zxystd I was having the same issue with ac9462 https://github.com/OpenIntelWireless/itlwm/issues/652, 
and just tried your file. It is working flawlessly now! Thanks.

kpostekk, 2021-07-23:
Works for me too. Connected quickly, no performance issues. Everything works fine.

[image]
[image]

JBramauer, 2021-07-28:
Is the fix merged into the latest alpha already or should I try the fix.zip from this issue thread?

zxystd, 2021-07-28:
@JBramauer It is merged now.

zxystd, 2021-07-28:
@kpostekk OK, thank you for the testing and feedback!

JBramauer, 2021-07-28:
@zxystd today's Alpha build is working fine on 9462 as well. Quite a speed improvement over 1.2.0 stable (which was the latest version working on 9462 prior to this fix). Thank you!

zxystd, 2021-07-30:
Fixed in 375d8a514b7772133e233135ab636a458e69447c, thank you all.

closed 2021-07-30 by zxystd