OpenIntelWireless/itlwm #630: [itlwm+HeliPort] ac8265 每隔几分钟就断连

JunkuiZhang, 2021-06-22:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v1.3.0 stable
- WiFi Card Model: ac8265
- PCI Product ID: 
- macOS Version: BigSur 11.4

**Description**
具体表现如下（以开机为起始）
- 开机后不会自动连接上次关机时连接的Wi-Fi（此Wi-Fi隐藏了ssid，且在更新HeliPort后出现，已经给了HeliPort相应权限）
- 手动输入网络连接后，过几分钟会出现网页刷不出的现象，短则十几秒，长则至少1min
- 通过HeliPort重新开关Wi-Fi后，网页仍有几率刷不出

**Bug Report Archive**
- 此log为Wi-Fi刚断连时生成：
[itlwm_logs_1.log](https://github.com/OpenIntelWireless/itlwm/files/6692023/itlwm_logs_1.log)

- 此log为上个log后第二次Wi-Fi断连时生成：
[itlwm_logs_2.log](https://github.com/OpenIntelWireless/itlwm/files/6692029/itlwm_logs_2.log)

- 此log为后续网络恢复后生成：
[itlwm_logs_reconnect.log](https://github.com/OpenIntelWireless/itlwm/files/6692112/itlwm_logs_reconnect.log)

**Kext Download Source**
下载于[release page](https://github.com/OpenIntelWireless/itlwm/releases)的Big Sur版本

zxystd, 2021-06-22:
日志里并没有看到掉线，你试试v2.0.0-alpha版本吧

JunkuiZhang, 2021-06-22:
> 日志里并没有看到掉线，你试试v2.0.0-alpha版本吧

非常感谢大神的回复，我晚上回去排查一下

JunkuiZhang, 2021-06-25:
把HeliPort从1.4.0降到1.3.1掉线的现象几乎没有了。。很奇怪

liy010, 2021-06-26:
[Log_2021-06-26_11-25-22.log](https://github.com/OpenIntelWireless/itlwm/files/6719515/Log_2021-06-26_11-25-22.log)
一样的情况，用的是最新的alpha版 AirportItlwm-Big_Sur-v2.0.0-DEBUG-alpha-93bf81f.zip，网卡型号 9260，连接的是 5G
使用1.3.1没问题 @zxystd

JunkuiZhang, 2021-07-06:
确定是heliport的问题，1.4.0版heliport经常掉线，换回1.3.1就正常了。
itlwm无论是最新beta版还是stable版在heliport 1.3.1版下均正常，在1.4.0下均会掉线。
再次感谢大神

zxystd, 2021-07-06:
@williambj1

williambj1, 2021-10-02:
@JunkuiZhang 能提供 HeliPort 的日志吗