OpenIntelWireless/itlwm #634: AirportItlwm驱动Intel 8265NGW网卡关联ap时无法协商到11ac协议

upkon, 2021-06-24:
主机硬件：ThinkPad x260
无线网卡：Intel 8265NGW，在Windows/Ubuntu/centos下测试正常。
Hackintosh版本：bigsur，11.4 (20F71) 
opencore：0.7
kext release：1.30 
问题现象：
1、无线AP的ssid单独关联至5G射频口时，8265网卡可以正常通过密码关联到这个ssid，也能工作，但是协商协议只能到11n，协商带宽为300Mbps（ap侧硬件射频口为2x2@866.6,Mbps，协议为11an 11ac自适应）。如果强制关闭ap侧的11an协议兼容，则网卡无法关联到这个ssid。
2、无线AP的ssid单独关联至2.4G射频口时，8265网卡可以关联到这个ssid，但是出现下行正常，上行严重丢包的现象。
已经尝试过的方法及结果：
1、修改ssid加密方式，无效
2、关闭ssid加密，无效
3、更换AP设备，无效
4、切换信道，无效，且通过omnipeek抓取空口报文，未发现有空口拥塞现象，11n的协商过程正常，未发现11ac协议交互。
附件截图：
这是关联到ssid后airport的状态
[image]
[image]
以下内容为ac上读取到的verbose内容，因涉及隐私，相关mac地址已打码
<H3C-WX2510H-AC01>display wlan client mac-address e4a4-7100-0000 verbose 
Total number of clients: 1

 MAC address                       : e4a4-7100-0000  
 IPv4 address                      : N/A
 IPv6 address                      : N/A
 Username                          : N/A
 AID                               : 2
 AP ID                             : 1
 AP name                           : 60da-0000-0000
 Radio ID                          : 1
 Channel                           : 149
 SSID                              : 4320H
 BSSID                             : 60da-0000-0000
 VLAN ID                           : 1
 Sleep count                       : 0
 Wireless mode                     : 802.11an
 Channel bandwidth                 : 40MHz 
 20/40 BSS Coexistence Management  : Not supported
 SM power save                     : Disabled
 Short GI for 20MHz                : Supported
 Short GI for 40MHz                : Supported
 STBC RX capability                : Not supported
 STBC TX capability                : Not supported
 LDPC RX capability                : Not supported
 Block Ack                         : N/A
 Supported HT MCS set              : 0, 1, 2, 3, 4, 5, 6, 7, 
                                     8, 9, 10, 11, 12, 13, 14, 
                                     15
 Supported rates                   : 6, 9, 12, 18, 24, 36, 
                                     48, 54 Mbps
 QoS mode                          : WMM
 Listen interval                   : 100
 RSSI                              : 0
 Rx/Tx rate                        : 0/0 Mbps
 Speed                             : N/A
 Authentication method             : Open system
 Security mode                     : RSN
 AKM mode                          : PSK
 Cipher suite                      : CCMP
 User authentication mode          : Bypass
 WPA3 status                       : Disabled
 Authorization CAR                 : N/A
 Authorization ACL ID              : N/A
 Authorization user profile        : N/A
 Roam status                       : N/A
 Key derivation                    : SHA1
 PMF status                        : N/A
 Forwarding policy name            : Not configured
 Online time                       : 0days 0hours 0minutes 40seconds
 FT status                         : Inactive
 BTM mode                          : Inactive
<H3C-WX2510H-AC01> 
另附上macOS的WLAN日志（相关mac已打码）：
[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/6710784/wifi.log)

zxystd, 2021-06-24:
1.3.0只支持上行聚合和11n的40mhz。
2.0.0支持了11ac，认真看更新日志。

upkon, 2021-06-24:
抱歉，由于我工作环境特殊，不能及时上外网。
貌似刚才我看了一下AirportItlwm  release还是1.30？
轮休才能回家上GitHub。我看下新的日志吧。

zxystd, 2021-06-24:
[image]

closed 2021-06-24 by zxystd