OpenIntelWireless/itlwm #638: Airportitlwm.kext takes too long to boot compared to itlwm + Heliport.

mifjpn, 2021-06-28:
Have You Read Our Docs
Yes

Are You Reporting A Bug
Yes

Environment　
    Kext Version: Airportitlwm.kext　Stable 1.3.0
    WiFi Card Model: Intel Wireless-AC 8260 Wireless Network Adapter
    PCI Product ID: XP3@1C,2/PXSX@0 pci8086,24f3    PciRoot(0x0)/Pci(0x1C,0x2)/Pci(0x0,0x0) 
    macOS Version: 11.4 Big Sur
Description
Itlwm + Heliport boots in normal boot time.
However, Airportitlwm.kext will wait a few minutes based on the following dmesg.

% sudo dmesg | grep Intel
[ 5529.957430]: [IGFB][LOG  ][DISPLAY   ] drmlog - doCmdAction calling clientNotify during sleep, client = kIntelClient_MEUserClient[IGFB][LOG  ][DISPLAY   ] clientNotify - Client: 1, Status: sleep, RC6Count: 0, fMediaReady: 1, fMediaCount: 1, Controller:Awake
[ 6404.259018]: PMRD: power clamp removed AppleIntelFramebuffer 10000038e, pendingCap 0xf, ps 2, cflags 0x2
[ 6404.259903]: PMRD: power clamp removed AppleIntelFramebuffer 10000038f, pendingCap 0xf, ps 2, cflags 0x9
[ 6404.259943]: PMRD: power clamp removed AppleIntelFramebuffer 100000390, pendingCap 0xf, ps 2, cflags 0x9
[ 6404.513768]: IntelFirmware: Driver Stop()
[ 6404.514987]: IntelFirmware: Driver free()
[ 6404.586950]: IntelFirmware: Driver init()
[ 6404.586970]: IntelFirmware: Driver Probe()
[ 6404.586986]: IntelFirmware: name=IOUSBHostDevice, class=IOService, vendorID=0x8087, productID=0x0A2B
[ 6404.587284]: IntelFirmware: Driver Start()
[ 6405.422551]: [IGFB][LOG  ][DISPLAY   ] doCmdAction calling clientNotify during wake, client = kIntelClient_MEUserClient
[ 6406.087509]: IntelFirmware: set configuration to 1
[ 6406.088081]: IntelFirmware: Found interface!!!
[ 6406.088120]: IntelFirmware: Found Interrupt endpoint!
[ 6406.088360]: IntelFirmware: Found Bulk out endpoint!
[ 6406.088572]: IntelFirmware: Found Bulk in endpoint!
[ 6406.088801]: IntelFirmware: usb init succeed
[ 6406.088806]: IntelFirmware: HCI_OP_INTEL_VERSION
[ 6406.089482]: IntelFirmware: interrupt wait
[ 6406.089834]: IntelFirmware: Bootloader revision 0.0 build 2 week 52 2014IntelFirmware: suspect device firmware: ibt-11-18-0.sfi
[ 6406.089856]: IntelFirmware: interrupt continue
[ 6406.089860]: IntelFirmware: HCI_OP_BOOT_PARAMS
[ 6406.090322]: IntelFirmware: interrupt wait
[ 6406.096845]: IntelFirmware: Device revision is 5
[ 6406.096854]: IntelFirmware: Secure boot is enabled
[ 6406.096857]: IntelFirmware: OTP lock is enabled
[ 6406.096860]: IntelFirmware: API lock is enabled
[ 6406.096864]: IntelFirmware: Debug lock is disabled
[ 6406.096867]: IntelFirmware: Minimum firmware build 1 week 10 2014
[ 6406.103736]: IntelFirmware: Found device firmware: ibt-11-5.sfi
[ 6406.103756]: IntelFirmware: interrupt continue
[ 6406.103759]: IntelFirmware: send firmware header
[ 6406.104207]: IntelFirmware: send firmware pkey
[ 6406.105144]: IntelFirmware: send firmware signature
[ 6406.109321]: IntelFirmware: send firmware data
[ 6407.498186]: IntelFirmware: boot_param=0x40800
[ 6407.498350]: IntelFirmware: send firmware done
[ 6407.498355]: IntelFirmware: interrupt continue
[ 6407.498359]: IntelFirmware: HCI_OP_INTEL_RESET
[ 6407.498524]: IntelFirmware: Intel reset succeed
[ 6407.498815]: IntelFirmware: Notify: Firmware download done
[ 6407.498822]: IntelFirmware: can not proceed response
[ 6407.498852]: IntelFirmware: interrupt wait
[ 6407.510097]: IntelFirmware: Notify: Device reboot done
[ 6407.510108]: IntelFirmware: can not proceed response
[ 6407.510124]: IntelFirmware: interrupt continue
[ 6407.510128]: IntelFirmware: HCI_OP_INTEL_EVENT_MASK
[ 6407.510487]: IntelFirmware: interrupt wait
[ 6407.519094]: IntelFirmware: interrupt continue
[ 6407.519118]: IntelFirmware: End download
[ 6407.519122]: IntelFirmware: Clean up...

I didn't understand why I got an error about "IOUSBHostDevice". Therefore, at first, I doubted the USB relationship and made various trials and errors. However, in the end, I arrived at the result of itlwm + Heliport.

Bug Report Archive

Kext Download Source
AirportItlwm_v1.3.0_stable_BigSur.kext.zip
itlwm_v1.3.0_stable.kext.zip

mifjpn, 2021-06-28:
sorry.
The problem cannot be reproduced well. The same thing started happening with itlwm, so I'll close it once.

closed 2021-06-28 by mifjpn

williambj1, 2021-06-28:
Your Bluetooth Kext is outdated, the latest Release Candidate should address your issue.

mifjpn, 2021-06-28:
Thank you
I try to intelBluetoothinject.kext(latest commit which is build myself).
Now I don't found issue.