OpenIntelWireless/itlwm #642: AC9560 使用 v2.0.0-alpha 22a83ab 连接不上wifi

caicaiget, 2021-07-08:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**

- Kext Version:   v2.0.0-alpha 22a83ab
- WiFi Card Model:  AC9560
- PCI Product ID: 
- macOS Version: BigSur 11.4

**Description**
使用最新的v2.0.0-alpha 22a83ab 会连接不上wif。
使用  v2.0.0-alpha 858c31a可以连接上

**Bug Report Archive**
[Log_2021-07-08_20-36-23.log](https://github.com/OpenIntelWireless/itlwm/files/6784664/Log_2021-07-08_20-36-23.log)

**Kext Download Source**
github

zxystd, 2021-07-11:
你确定？有没有再换回最新的测试一遍看看

caicaiget, 2021-07-12:
测试过了，采用目前最新的kext会连接不上，状态是wifi已连接，但是无法连接互联网，并且没有ip地址。

zxystd, 2021-07-12:
已连接跟你说的这个问题是两码事

caicaiget, 2021-07-12:
好的，请问有排查到是什么问题导致的吗

zxystd, 2021-07-12:
3e63230703f11d6daf13f5fa69d6e77a65b8cf3c 这个提交修改了连接状态，但是内测群的群友都测试过了没有问题（mac10.15，mac11，mac12），而且可以正常DHCP。

caicaiget, 2021-07-12:
有没有用AC9560测试呢，或者还需要我提供日志配合调查。

caicaiget, 2021-07-12:
[image]
刚开机会遇到zh这个情况。我换了个wifi就好了。

caicaiget, 2021-07-12:
我刚刚测试，开机启动，连接5Gwifi就会这样，有ip地址，无法连接互联网，切换为2.4G就好了就可以连接成功。
[image]

caicaiget, 2021-07-12:
感觉像是开机后的首次连接会有问题，可以通过切换wifi解决

zxystd, 2021-07-12:
内测群

1132224334

caicaiget, 2021-07-12:
申请了，同意一下。

closed 2021-07-31 by caicaiget