OpenIntelWireless/itlwm #643: AX200 - Itlwm reconnects every about 10 minutes

khanhtran-cse, 2021-07-11:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v2.0.0-alpha (and both 1.3.0)
- WiFi Card Model:  AX200
- PCI Product ID: Unknown
- macOS Version: 11.4

**Description**

Wifi disconnect (about 10s-15s) and reconnect every 10 minutes.

**Bug Report Archive**

[bugreport_39781.zip](https://github.com/OpenIntelWireless/itlwm/files/6797472/bugreport_39781.zip)

**Kext Download Source**

Download from Release page

zxystd, 2021-07-12:
Please check your router setting to see if the Pairwise key expired time is 10min.

khanhtran-cse, 2021-07-12:
I don't see any config has key Pairwise on my router.

williambj1, 2021-07-12:
What is the model of the router? Does is disconnect on other routers or hotspots?

khanhtran-cse, 2021-07-12:
I tested on two router:
1. ISP Router: disconnect every 10 minutes
2. Xiaomi 4A Router (connect to ISP Router): work fine.

zxystd, 2021-07-24:
@khanhtran-cse Airport seems to have key updating problem, if the router configure as periodic key expires may have problems, so `itlwm.kext` is recommended, can you try that?

khanhtran-cse, 2021-08-16:
I found the reason that make kext disconnected is ISP Router was set to leave IP every 10 minutes (DHCP leave time). Thank you @zxystd.

closed 2021-08-16 by khanhtran-cse