OpenIntelWireless/itlwm #644: a mistake in Mojave with OC v0.6.0

pzw1999, 2021-07-12:
yesterday I set macos 10.14.6 on my matebook  which version is 2019,the wifi wasn't work.i can not notice the wifi ico in the status bar.Network setting has not wifi option,either.Airportitlm version is 1.3 on Mojave,i do not know what shohld i do next.Appreciate for your help.
ps. matebook hardware list:
CPU:i3 of 8th
RAW:8g
GPU:UHD 620
networkcard:9560
hard disk:rc500
[image]
[image]
[image]

zxystd, 2021-07-12:
Works fine on Mojave.

closed 2021-07-12 by zxystd

pzw1999, 2021-07-12:
clover or OC?